supermerill/SuperSlicer #243: First layer not adhering at all but skirt is fine

HCR-Mad-Bull, 2020-05-14:
2.2.49 Slic3r++

Windows 10 version 1909

Anycubic Predator 

Hi, when I try to print any files the first layer and brim refuses to lay down at all but the skirt is perfect. I'm not sure what else to try. Tried a few different settings but nothing helps. Other slicers work fine.

The layout and the settings seem very nice and prefer it over the others.

Thanks Ben

supermerill, 2020-05-14:
Best way to make it stick is go go slow on first layer (see in speed tab) and thick (like a 140% width, see in width tab).
If it's pla, you can spit on your build plate and wipe with a cloth.
i'll add some tuning help in next release, stay tuned.

HCR-Mad-Bull, 2020-05-14:
Hello, thanks for answering but I just figured it out, it was actually severally under extruding by 30%. Really odd as the skirt seemed was fine.  Thanks again, not too often the developer personally responds.

closed 2020-05-14 by HCR-Mad-Bull