supermerill/SuperSlicer #246: [Feature Request] Brim extrusion width

stevenignys, 2020-05-18:
Version: Slic3r++_2.2.50.0_win64_20200510 (Windows 10)

Would it be possible to have a extrusion width option for the brim. My brims seem very thin and not sticking which is making them useless for the tall prints with small bottom footprints I'm trying to print. 

I'm using a 0.4mm nozzle with extrusion width 0.5mm on 0.2mm layer and the parts are fine but the brim is not?

[image]
[image]

supermerill, 2020-05-18:
It's the same process, flow and width, how can the first layer can be good on the part and bad in the brim?

liftbag, 2020-05-18:
Thinking about how I could replicate such behavior with my printer (which instead prints correctly brim and first layer), I think I should reduce the flow rate, either with a bad setting of the extruder steps/mm, or with a bad (or missing) setting of the extrusion multiplier. It wouldn't be enough, I think, I should also increase the z-offset of the first layer.
Although little highlighted by the photo, I seem to see the same problems as the brim on the first layer of the piece. Slic3r ++ has a separate setting for the the first layer flow. You could act on that, as long as that's not the problem, like it's set too low.

stevenignys, 2020-05-27:
Issue appears to be incorrect z0 height.

closed 2020-05-27 by stevenignys