supermerill/SuperSlicer #248: [Feature Problem] Ugly "Reversed" overhangs with one perimeter on.

MarcPot, 2020-05-19:
### Slic3r++ 2.2.50
### Windows 10 (build 18363), 64-bit edition
### Behavior
When using Print Settings -> Perimeter & Shell -> Quality (slower slicing) -> Only one perimeter on Top surfaces while there is an reversed overhang (hill) in your model.

When there is an overhang like this the sliver will act very strange there, it adds one perimeter, then a bit of top layer and then your actual walls. This only happens when the slope has a low angle
I hope the image will explain the problem enough.

This problem results in a very weak and ugly outer shell on reversed overhangs.

These pictures are without Only one perimeter on Top surfaces:
[image]
[image]

These pictures are with Only one perimeter on Top surfaces:
[image]
[image]

#### Project File (.3MF) where problem occurs
This is an example, but it occurs with every model like it.
[z_drive_main_a_x2.zip](https://github.com/supermerill/Slic3r/files/4650312/z_drive_main_a_x2.zip)

supermerill, 2020-05-19:
It detects that there is an area that can be laid out with the top pattern, so it's doing its job. To be able to have enough space to do the pattern, it's grown by "anchor solid infill by X mm". 
It is really a problem?

You can remove "only one perimeter on top for these layer where you don't want it (or only where you want it, as you wish) with the height range modifier (or modifier mesh, but they can add seams when working on perimeters).

Alternatively, you can change the top pattern in a modifier mesh, set it to "concentric (filled)" which is almost the same as perimeters, and put the modifier where you don't want the hilbert pattern.

It is enough to resolve your problem?
I don't think it's possible to know where you want top infill and where you don't if you don't put some modifier to explain your tough to the program, but maybe you have an idea on how i can be implemented.

MarcPot, 2020-05-19:
Yea you're right, it's probably just easier doing it by hand. 

Only thing I can think of is a setting like "Minimum distance to next layer perimeter" that looks where the single perimeter are closest and if that distance is to low the single perimeter wouldn't be used. But since I'm no programmer, I have no idea how hard it would be to implement.

supermerill, 2020-05-19:
It may be possible to have a setting like that.
"minimum top thickness for infill" ?
in mm or perimeter width %

MarcPot, 2020-05-19:
Yea like that, would be awesome! Even less reason to turn the function off.

With hilbert curve and one perimeter the perimeter almost gets invisible, would be awesome if it works at even more parts!

MarcPot, 2020-06-29:
@supermerill Just checked out the fix, and it's working exactly as I meant. Awesome work!!!

closed 2020-06-29 by MarcPot