supermerill/SuperSlicer #249: Support interface pattern dropdown box visual artifact

liftbag, 2020-05-20:
### Version
_2.2.50 commit f1c6cc602d2f97f5e454771c8086de3c4882c673_

_adding the temperature calibration (wip, had to add some pictures)._

### Operating system type + version
_win10_

### Behavior
* _Don't know if it's pc related but the dropdown box appears as follows on y computer. Either way, it works properly._
[image]

Staying on the subject, wouldn't it be preferable to start the temperature tower with the higher temperature? If the extruder become clogged due to a too low temperature, it would be better to happen on top than at the bottom.
Congratulations on the great, really great job on slic3r ++ supermerill.

supermerill, 2020-05-20:
I happens also on my computer sometimes. I don't know why but it's maybe due to a weird interaction with layouts when the size is near the max. May be hard to diagnose.

> wouldn't it be preferable to start the temperature tower with the higher temperature? 

i was wondering what is the best option between the two, tanks for the input.

liftbag, 2020-05-20:
Could the temp tower be customizable? For example, establish the start, end and step temperatures.

liftbag, 2020-05-20:
Another thing. Looking at the Z slider in the temp tower slicing preview, I don't see the temperature changes. Will they have to be entered manually, or when the project will be finalized, the temperature changes will be automatically entered by the configuration assistant?

liftbag, 2020-05-20:
The issue is not present in the build macos #68: Commit aea2f3b.
However, oddly the calibration menu does not work, producing the following error when starting the assistant (whatever it is).

[image]
Maybe _Contents_ is missing in the path?

supermerill, 2020-05-20:
> Could the temp tower be parameterized? For example, establish the start, end and step temperatures.

The start can be parametrized, as the middle tempis the one you ahve in your current filament setting. I can do better, like adding an input to the number of previous & next temp, and the step, 

But the numbers are pre-made and only available for multiple of 5 starting by 180.
For the step input, maybe i can restrict to 5/10/15/20 ? do you think it's useful to have a 2.5 degree step?

For now i have to print some, test and write some explanations.

> I have just checked the gcode and verified that the temperature changes are already correctly entered. Unfortunately, the preview extrusion temperatures cannot be visually checked.

yeah, i was lazy and just modify the after layer gcode sneakily. (you have to reset it if you want to print somethgin after it.
I forgot about the Z slider, will change for that.

> However, oddly the calibration menu does not work, producing the following error when starting the assistant (whatever it is).

This feature think that the resources directory is inside the slic3r executbale directory (current location). It works in windows, but maybe it's different in macos, true. Have to search it instead of blindly expect it, and display an error if not found.

liftbag, 2020-05-20:
> For the step input, maybe i can restrict to 5/10/15/20 ? do you think it's useful to have a 2.5 degree step?

2.5 degree step is useless in my opinion.

> I forgot about the Z slider, will change for that.

Ok.

liftbag, 2020-05-20:
> This feature think that the resources directory is inside the slic3r executable directory (current location). It works in windows, but maybe it's different in macos, true. Have to search it instead of blindly expect it, and display an error if not found.

But the executable directory has those resources. The path is: /Applications/Slic3r++.app/Contents/resources/calibration/

supermerill, 2020-05-20:
prusa has already made the util function to get the good path, i'll use that.

supermerill, 2020-05-20:
all that (but the first issue) should be on this nightly build.

liftbag, 2020-05-21:
> all that (but the title issue) should be on this nightly build.

Wow! Everything works! Thanks :)

liftbag, 2020-05-22:
Hello @supermerill: the resources for the Bridge pattern tuning and Ironing pattern tuning cannot be loaded again on macOS version of the latest nightly "external_perimeter_cut_corners setting".
This time the wizard is correctly loaded, but not the models to generate the objects for testing.

[image]

supermerill, 2020-05-22:
erf, a bad copy-cut made before the fix.. easy fix.

liftbag, 2020-05-23:
> erf, a bad copy-paste made before the previous fix.. easy one.

Bridge pattern tuning still need to be fixed, on macOS and win10.

liftbag, 2020-05-23:
In win10 I managed to load the Bridge pattern tuning after deleting the configuration folder, starting from scratch and loading the saved config bundle.

supermerill, 2020-05-23:
what's the errors you had?

liftbag, 2020-05-23:
> what's the errors you had?

I understood why I had errors that seemed random to me. It only happens when you want to load the bridge flow test with 2.5 steps.

[image]

closed 2020-06-24 by supermerill