supermerill/SuperSlicer #250: Error while exporting config-bundle and using a printer with Klipper and firmware retraction

bigsubgit, 2020-05-20:
### Version
2.2.49

### Operating system type + version
macOS Catalina 10.15.4

### 3D printer brand / version + firmware version (if known)
A custom ender3 with klipper fw

### Behavior
If you having a klipper 3d printer and you use firmware retraction, when you export config-bundle.ini you have this error.
The problem is only exporting config-bundle, while the use of G10/G11 command is correct; they are correctly used on g-code files exported.

[image]
[image]

supermerill, 2020-05-20:
i'll add klipper on the list of authorized firmware.

bigsubgit, 2020-05-21:
Hello, @supermerill the new build works correctly. 
Many thanks.

closed 2020-06-24 by supermerill