supermerill/SuperSlicer #251: Cutting corners seems broken

liftbag, 2020-05-22:
### Version
_2.2.50 nightly build " external_perimeter_cut_corners setting" (Commit 77468ba)_

### Operating system type + version
_macOS Catalina 10.15.4_

### Behavior
* _I was curious to test the new feature, but it seems broken_
* _Steps needed to reproduce the problem_
  * _enter any non-zero value in "Cutting corners"_
  * _hit Slice now_
* _Expected Results_
  * _it's not clear to me, it's a new feature. I imagined to have a normal slice, and to be able to visualize the changes in the corners with the "volumetric flow rate view"_
* _Actual Results_
  * _The slice is broken into multiple areas on top layers. Vertical corners are cutted, but not homogeneously. Nothing seems to change by changing the value in "Cutting corners", just enter any value other than zero to always get the same result._

Cutting corners set to 0
[image]

Cutting corners set to 1%
[image]

Cutting corners set to 100%
[image]

#### Project File (.3MF) where problem occurs
[xyzCalibration_cube.3mf.zip](https://github.com/supermerill/Slic3r/files/4670483/xyzCalibration_cube.3mf.zip)

liftbag, 2020-05-22:
I had the doubt that, somehow, the new feature exalted a mesh problem, but tried with another mesh, the result is similar.

[image]

supermerill, 2020-05-22:
yeah, i didn't had the time to test it, i just pushed it to be able to compile it on an other pc.

supermerill, 2020-05-23:
should be fixed, but not tested yet. Also the fail macos build is a lie.

liftbag, 2020-05-23:
> should be fixed, but not tested yet. Also the fail macos build is a lie.

I'll wait for your test, because I didn't see any change when enabled.

supermerill, 2020-05-28:
Well, you need a big nozzle and very precise flow to be able to see anything. It's a 5% flow difference on a very little distance.

closed 2020-06-24 by supermerill