supermerill/SuperSlicer #252: Sequential printing bugs report

liftbag, 2020-05-24:
### Version
_2.2.50  Commit 572910e_

### Operating system type + version
_macOS Catalina 10.15.4_

Sequential printing has some problems, most of which are not in PrusaSlicer, I will try to list them separately, although it is hard for me to find a certain repeatability.

### 1. Auto arrange
#### Behavior
* This one is derived from PrusaSlicer, although it's more of a nuisance than a bug. When sequential printing is selected, the self-arrangement of objects on the bed places them too close together, generating an error. You have to move them manually in order to complete the slice.
* Steps needed to reproduce the problem
  * _Skirt and brim disabled, load two or more objects. If they are too close, according to the sequential print settings, a warning message is shown_
  * _Hit Arrange button_
* Expected Results
  * _Objects should be placed at an adequate distance from each other which prevents collisions with the extruder._
* Actual Results
  * _The following error is shown and slicing is prevented for invalid data. By manually spacing the objects, even very little, it allows you to complete the slice._
[image]
Note: I have specified skirt and brim disabled, because enabling them you can have a series of problems that I will try to describe below.

### 2. Sequential printing takes brim and skirt into account in collision prediction.
#### Behavior
* This doesn't happen in PrusaSlicer. 
* This is easily verifiable, observing the necessary spacing of the objects to avoid collisions with the extruder in sequential printing: it changes according to the thickness of the brim or the distance of skirt (when individual skirt is used).
I think it should be corrected as it fulfills a useless precaution, as it is impossible for the extruder to collide with the brim. It could collide with the skirt if Draft shield is selected, but this setting is ignored when sequential printing is enabled.

### 3. Relocation issues when skirt and/or brim are enabled.
#### Behavior
* I tend to believe that this problem is perhaps attributable to my computer, it should also be verified on other software platforms: after a successful first slice, a simple rearrangement or repositioning of the objects on the bed (when skirt and brim are active) leads to the freezing of the program to the next slice._
* Steps needed to reproduce the problem
  * _Set skirt and brim on_
  * _Load an object_
  * _Copy/paste the object_
  * _Close the warning message and hit auto arrange button_
  * _Hit slice button_
  * _Go back to the 3D editor view and move one of the two objects. If you approach them, select auto arrange again._
  * _hit slice button again_

* Expected Results
  * _The slice should be reprocessed_

* _Actual Results_
  * _Slic3r ++ freezes in processing the gcode and must be forcibly terminated_
#### Project File (.3MF) where problem occurs
[MR126ZZ_eliminator_x_5.3mf.zip](https://github.com/supermerill/Slic3r/files/4673281/MR126ZZ_eliminator_x_5.3mf.zip)

### Auto arrange
#### Behavior
* _This one is derived from PrusaSlicer, although it's more of a nuisance than a bug._
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Hit slice button again_

There are other inconsistencies, but it is not easy to identify their repeatability. If appropriate, I will report them in future comments to follow.

liftbag, 2020-05-24:
I just checked that issue no.3 also happens in win10.

supermerill, 2020-05-24:
1. erf, not again.
Also this pop-up is annoying as fuck, i'm going to replace it by a warning in the background.
2. as you print a skirt & brim per-object in my fork, it's needed. I may have forgotten to remove this verification if "only one skirt" is selected.
3. noted

liftbag, 2020-05-24:
>   2. as you print a skirt & brim per-object in my fork, it's needed. I may have forgotten to remove this verification if "only one skirt" is selected.

But also PrusaSlicer print the brim per object, but it is not taken into account about collisions in sequential printing. Only the objects are included in the calculation of the safety distances.
It's possible that few feel the problem. I have a relatively small bed with a bulky extruder. On some occasions I realize that printing multiple items that need a very thick brim (for example using polypropylene), limits the possibility of adding objects compared to what I can do with PrusaSlicer, which excludes the brim from possible collisions with the extruder.

>  Also, maybe you want a "only one brim" setting"?

I didn't understand the question. I certainly wouldn't want only one brim for all objects in sequential printing.

liftbag, 2020-05-24:
I had one report about the auto arrange poin one: seems that on linux environment the pop-up is not shown.

supermerill, 2020-05-24:
if the brim isn't taken in to account, you can touch the other object when printing it.

supermerill, 2020-05-30:
I'm going to consider the skirt as part (like an ooze shield) of the object if it's taller than 3 layer (and if complete_objects_one_skirt isn't checked).

Some problems that may arise and aren't checked right now:
 - if the brim width is modified for only one object and is really big, it can make the extruder collide to other as i check only the general config brim size (right now).
 - if there is a tall skirt and complete_objects_one_skirt , it's going to wreck havoc unless you print it really far away from parts.

liftbag, 2020-05-31:
Thinking back, during the execution of the brim the risk of collision must be taken into consideration, and currently, I seem to understand that PrusaSlicer does not consider the brim in the spacing of objects in sequential printing.
To be sure and optimize the distances, we say that the already extruded brim should not be taken into account in the spacing of the next object, and so on. Is this correct?
In the meantime I found a situation of sure collision due to the brim assigned to a single object. I will describe it to you as soon as possible.

liftbag, 2020-06-02:
@supermerill  I don't know if something was done with version 2.2.51, but I can't reproduce the anomalous behavior that I had noticed with version 2.2.50. Now sequential printing takes into account the already extruded brims by excluding them from collisions. Instead, during the current brim, the extruder will never collide with already printed objects.
Issue no.1 has been fixed, and even issue 3 I can no longer reproduce with the superslicer edition.
If you deem it appropriate, it can be closed for me.

If I can afford it, a note on the new name. It is clear that it refers to your nickname, however I preferred the old name, which implied a version of slic3r on steroids. I am part of an Italian PrusaSlicer/Slic3r++ telegram group. Almost all the active members passed from PrusaSlicer to Slic3r++ and there is a general perplexity about the new name and more or less everyone preferred the old one.
Anyway, good job with SuperSlicer. It is quite amazing to see how active development is, especially when compared to softwares that have a large development team.

supermerill, 2020-06-02:
Erf, it's almost too late for the name.
If only i had a way to launch a fair and wide poll...

liftbag, 2020-06-02:
> Erf, it's almost too late for the name.
> If only i had a way to launch a fair and wide poll...
> some people hate the '3', some love it...

It's not really important. What matters is the improvement policy that you manage to implement. I believe very few can feel frustrated in their expectations by using SuperSlicer.

supermerill, 2020-06-02:
to get back on topic, i changed the way the checking algorithm works, so now it should be good to detect problems, but it's not smart enough to arrange them in the most optimal way.
Also, i didn't had the time to check how object-modifiers for brim may create problems, so it's still a wip things.
And you may encounter a bug where an object jump at (0,0), but goes back at the right stop on the next slic click.

liftbag, 2020-06-02:
> to get back on topic, i changed the way the checking algorithm works, so now it should be good to detect problems, but it's not smart enough to arrange them in the most optimal way.

Yes the automatic arrangement works a little unpredictably, but by manually positioning the objects, the collision prediction works as I would expect.

closed 2020-06-24 by supermerill