supermerill/SuperSlicer #253: Unwanted solid infill causing bad overhangs

PuddingBaer91, 2020-05-24:
Hi,

I found out that if you set a high number for bottom layers (for example 10) then the whole model will be filled at some places with solid infill. This is causing bad overhangs.

[image]
[image]

[image]
[image]

[image]
[image]

As you can see this is sliced with Ideamaker but it happens on all the other slicers also. If you set bottom solid layers to 5 both overhangs will print bad. If you set it to 1 only one side will be bad because Ideamaker adds some more solid fill to the left side.. (May Slicer handels this in another way)
If you set the bottom layers to 0 the overhangs will printed fine because no solid fill at the overhang section.
For me i only want solid bottom layers beginning from z=0. For example if you set solid bottom layers to 5 it should stop with them after the 5 FIRST layers

supermerill, 2020-05-24:
can you post the project file? (.3mf) zipped here?

PuddingBaer91, 2020-05-24:
Yes i can. I will so later if i am on my pc. In the cura discussion i mentioned above we found the solution for cura. In cura you are able to set initial bottom layers and bottom layers. If you set bottom layers to 0 and initial bottom layers to ur desired value it will work as i want.

64Florian, 2020-05-24:
For @supermerill 
-> This thread was discussed in Ultimaker/Cura#7817

supermerill, 2020-05-24:
If you want a manual solution, you can create a layer modifier and set solid bottoms layers to 0 in it.

PuddingBaer91, 2020-05-25:
yeah good idea with modifier but may you can do it like cura with **initial** bottom layers option
project file (.3mf) zipped:
[Extrudr-Bunny2017-No_Sunglass.zip](https://github.com/supermerill/Slic3r/files/4675387/Extrudr-Bunny2017-No_Sunglass.zip)

supermerill, 2020-05-25:
can you try to print it using "concentric (filled)" as solid pattern?

supermerill, 2020-05-28:
As there is a solution with the layer modifier, as the proof is not made with this slicer (or i misunderstood), i won't do anything until it's proven that it's not reproducible with this slicer (and it's not an obvious configuration problem).

supermerill, 2020-10-13:
@VanessaE 

> Put it back the way it used to be in the old days

Do you have a commit or a date on slic3r where it was the good "old days"? To compare with the current code / behavior.

VanessaE, 2020-10-13:
> Do you have a commit or a date on slic3r

Unfortunately no.  All I can remember is it was a few years ago.  But I'm 90% certain this behavior changed when the "ensure top/bottom thickness" feature was added.

> > Also you're a bit difficult to follow:

> That would be fine if it were adding more perimeters to fill it out.
> The solution is just to not add any extra plastic,

By this I meant, if you add any plastic at all, it has to be in the form of lines drawn parallel to, and going in the same direction as the perimeters, whether that be in the form of real perimeter loops or just simple lines that have an equivalent effect.

But if that can't be done -- even if you can't even just "fake it" -- then it's better to just not add this extra material at all, in any form.  Let the user add more perimeters in their config if they want.

jesterhead82, 2021-04-22:
This gets triggered either by "Ensure Vertical Thickness" to ensure perimters and also "Top/Bot
tom minimum thickness"

Would it be possible to set a different infill type (e.g. concentric fixes this) when the slicer decides to add solid infill to ensure vertical/horizental width?

VanessaE, 2021-04-22:
> This gets triggered either by "Ensure Vertical Thickness" 

No, as has previously been wrung out, it does not have to do with this.

PuddingBaer91, 2021-04-22:
@VanessaE please try to set a minimum layer time too slow down your print speed. Or just print slower to test your overhang performance. The slower you print the more time the filament has to cool and harden...

closed 2021-04-22 by PuddingBaer91

supermerill, 2021-04-25:
discussion about this issue can continue on #1090