supermerill/SuperSlicer #254: First layer height in percent is bugged

th33xitus, 2020-05-24:
### Version
2.2.50

### Operating system
Win10

### Behavior
I encountered a weird behavior when using values in percentage for the first layer height.
[image][image]

liftbag, 2020-05-24:
I think it's the normal behavior as described in the tooltip. The percentage is about the nozzle diameter. Assuming you have a 0.4 nozzle, by setting the first layer to 100% mean that is 0.4mm, which is a lot bigger than the set layer height, which is 0.1mm here.

th33xitus, 2020-05-24:
I see, thanks for the headsup. I usually use PrusaSlicer and I didn't read the tooltip.
Im wondering why this parameter is changed completely in its function when compared to PrusaSlicers first layer height. There its calculated over the default layer height, so no wonder i was expecting a different result when setting it to 100% in Slicer++.
Actually i don't get the idea of why you would ever calculate the first layer height over the nozzle diameter. Makes no sense to me. Makes things just unnecessary complicated and confusing.

supermerill, 2020-05-24:
because that's what dictate the first layer height.
And i don't want it to change when i modify the layer height.

closed 2020-05-24 by supermerill

th33xitus, 2020-05-24:
> And i don't want it to change when i modify the layer height.
That makes no sense at all. If you don't want the first layer height to change simply put in a value in mm? Then it's fix and you can still change the default layer height. But making it dependent on the nozzle is just weird. But okay. Your fork, your rules.

supermerill, 2020-05-24:
It's dangerous to let it be a % of the layer height. If you have it to 66% and then change you layer height from 0.3 to 0.1, then you may damage your bed.