supermerill/SuperSlicer #255: Unable to choose second extruder for supports (locked on extruder 1)

MarsArtis, 2020-05-27:
### Version
Slic3r++2.2.47 and 2.2.50

### Operating system type + version
MacOSX 10.13.6

### 3D printer brand / version + firmware version (if known)
Anycubic Delta Linear + Flying extruder and 2 extruders

### Behavior
Print settings/ Multiple extruders: the Support materia/raft interface extruder is always greyed out, locked on **extruder 1**
Printer settings/support material/genearate support material enabled does not make the previuos option unlock (while this happens in PrusaSlicer, I mean I can choose second extruder for supports)

[image]
[image]

So It's the support material extruder dependand on any other option I'm not aware of?

#### Project File (.3MF) where problem occurs
[nozzle-wrench-FABER_DD.3mf.zip](https://github.com/supermerill/Slic3r/files/4688193/nozzle-wrench-FABER_DD.3mf.zip)

neophyl, 2020-05-27:
You have to make your Interface layers at least 1 in Support material>Options for Support material Interface>Interface Layers for the Support Material/Raft Interface extruder to actually be used.

MarsArtis, 2020-05-27:
thx for pointin that. It's now selectable.
Just wondering why this have to be enabled to let extruder selectable

closed 2020-05-27 by MarsArtis

neophyl, 2020-05-27:
Basically same reason you have to have support enabled to change those settings.  The one that wasnt selectable was the support interface extruder and with the interface layers set to 0 its basically not used.  I agree in this case that it wouldnt make much difference, probably the same gui rules from forever ago.