supermerill/SuperSlicer #256: First Layer Speed modifier partially ignored

kizza42, 2020-05-29:
### Version
2.2.50

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Tevo Little Monster with Klipper

### Behavior
It seems the First layer speed modifier is partially ignored in this version, The brim prints at this speed but first layer of the model does not. This issue does not present in PrusaSlicer
[image]
[image]

#### Project File (.3MF) where problem occurs
[Extrusion_Test.3mf.txt](https://github.com/supermerill/SuperSlicer/files/4700323/Extrusion_Test.3mf.txt)

supermerill, 2020-05-29:
This only happen if you let the auto-speed algo do the job. Did you let some speed box to 0 (perimeters, solid infill,...) in prusa also?

kizza42, 2020-05-29:
The same settings in Prusa seem to honour the first layer speed.

[image]
[image]

commit 493dec83 (2020-05-30): fix #256, set auto speed before applying the first layer one
fix string type for tab.cpp
closed 2020-05-30 by supermerill