supermerill/SuperSlicer #257: Not working on MacOS Volume with Capital Letters

rotor5, 2020-05-30:
### Version
2.2.50

### Operating system type + version
MacOS 15.5.5 with APFS with capital Letters

### 3D printer brand / version + firmware version (if known)
Duet3d but does not matter

### Behavior
If i put the slic3r++ app into my App Folder it will not open.
(When I use it out of the dmg File it is working.)
And on my old MacBook with a MacOS Extended Volume it is working.

If I put the File on an USB Drive with FAT, it is working.

**For me it looks like the only Filesystem that makes problems is APFS with capital Letters.**

supermerill, 2020-05-30:
As i'm changing the name, i'm doing a pass on these things.
I'll use SuperSlicer for everything but the exe.
If it won't work, i'll change that to have a a lower-case exe only for linux.

supermerill, 2020-06-03:
Can you test the latest release?

rotor5, 2020-06-03:
I just tested it.
But I have the same/similar behavior as before.
The first thing i see when I put it out of the *.dmg is, that the icon is missing and it is not starting.
But now I have the question from MacOS if I am sure that I want to open... But the slicer is not opening.

supermerill, 2020-06-03:
I think I've use the same case for everything for this macos build...
i'll check again but i don't know what do to.

DeviousPenguin, 2020-06-03:
For what it's worth I'm SuperSlicer latest build works fine for me on MacOS 10.15.4, but my AFPS volume name is all lowercase

commit b564b17e (2020-08-26): #424 use Resources instead of resources

maybe also fix #257
closed 2020-08-24 by supermerill