supermerill/SuperSlicer #258: Bottom solid layer setting ignored if using height range modifier for 1st layer

jakep82, 2020-05-31:
### Version
2.2.50

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Klipper

### Behavior
Using a height range modifier to force a single perimeter for the first layer causes the slicer to ignore the bottom solid layer setting.  Without a modifier I see 5 solid layers as expected.  With a modifier for 0-0.2mm to use a single perimeter I only get solid infill on the first layer, and subsequent layers have standard infill.  I see the same behavior in Prusaslicer 2.2.0

[image]
[image]

supermerill, 2020-05-31:
can't reproduce, maybe you should post your project file (zipped)

here is my test, that is working: 
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/4707623/Shape-Box.zip)

jakep82, 2020-05-31:
I imported your project and used your settings to slice the box.  That seemed to work fine.  Adding AB drive parts for a Voron 1.8 and using the same height range modifier setting resulted in a hard crash.  I've attached a project file using my settings that doesn't crash, but does produce only a single bottom solid layer.  I'll spend some more time going back and forth to see if I can figure out what might be causing this difference in behavior.

[B_drive.zip](https://github.com/supermerill/SuperSlicer/files/4708306/B_drive.zip)

jakep82, 2020-05-31:
I found the setting that causes the issue.  With ensure vertical wall thickness enabled I get 5 solid bottom layers as expected.  With that setting disabled I only get 1 solid bottom layer.

supermerill, 2020-05-31:
Think it's something about the region, that 'bottom' doesn't transmit the modifier barrier but that option act as a failsafe here.

jakep82, 2020-06-01:
Thanks.  For now I'll just make sure I leave the ensure vertical shell thickness setting enabled, and I'll use the height range modifier.  I'm getting excellent looking prints with the single outline top and bottom.  Ironing is also pretty amazing.