supermerill/SuperSlicer #260: Feature request - modify the Marlin output to avoid M20X code at the beginning and more

onekk, 2020-05-31:
### Version
2.2.50

First of all, many thanks for this great software, I'm an user of slic3r (plain) and I'm pleased to see that this fork of PrusaSlicer is better than the original Slic3r.

Some feature request:

1) It will be possible to add a checkbox to avoid the M20X prints to set the speeds and accelerations at the beginning of the gcode, it was asked some times ago in PrusaSlicer but it was rejected with the "partly wrong" answer that these code were Marlin standards (no other slicer output these code, when selecting Marlin flavour)

2) Add an input box because is a pain to find an exact rotation, like 90, 180, 270, and so on.
I see that there are page up and page down, to rotate 45° CW and CCW maybe adding two button on the top button row like in the original slic3r will suffice.

3) when deselecting the fans, some fans are activated on the top layer, maybe a checkbox that permit to deactivate all "material fan" will be a good choice.

Thanks again for the good work.

Regards and Salut

Carlo D.

supermerill, 2020-05-31:
1.  it's not enforced on klipper, because they can use their own start_macro. What is the use-case where you have to print without heating the extruder as defined in the filament tab? Maybe you want that these gcode are not printed if the filament temperature is 0 ?

2. pass on expert mode and you have one in the right panel. Also, there is snap-in white lines just hover the; when rotating.

3. i don't understand. what is "deselecting the fans" ? what are "material fan"? can you be more precise, giving a project file (save as, then zip then drop here) and expected behavior?

happy printing!

onekk, 2020-06-01:
>     1. it's not enforced on klipper, because they can use their own start_macro. What is the use-case where you have to print without heating the extruder as defined in the filament tab? Maybe you want that these gcode are not printed if the filament temperature is 0 ?
> 
I have not cited temperature problem, only the M20X set for speed and acceleration that are put in the gcode at the star of the file.

>     2. On expert mode and you have one in the right panel. Also, there is snap-in white lines just hover them while rotating.
> 
 Yes I see the preset  on the line at right, thank you.

>     3. i don't understand. what is "deselecting the fans" ? what are "material fan"? can you be more precise, giving a project file (save as, then zip then drop here) and expected behavior?
> 

OK sorry in the "Filament settings" under "Cooling" there is section "Enable" and the second section "Fan Settings", If I disable the fans in the Enable, the setting in Fan Settings are kept, (for "Bridge fan speed" and "top fan speed"), even if in the Enable section there is a remind "fan will be turned off", this lead to the wrong assumption that the fan is not used.

For "material fan" I intend the fan that cools the extruded material.

More generally there are a HE fan than cools the Hot End, that is controlled by the firmware.
A Material Fan that cools the extruded material, and generally is controlled by the M106 M107 codes

```
* M106 - P<fan> S<speed> F<frequency> U<pin> L<min speed> I<inverted logic> H<int> Set Auto mode - H=7 for controller - H-1 for disabled
 * M107 - P<fan> Fan off
```

And maybe an electronics fan that cools the control board and the drivers, controlled by the firmware, but in MK4Duo is controlled even using the M106 and M107 extended form

> happy printing!

Thanks

Carlo D.

supermerill, 2020-06-01:
1) my bad, i'll add a switch in the machine limits tab.

3) understood, i'll make a pass on these, i didn't do a good enough job when i added these two boxes.

onekk, 2020-06-01:
>     1. my bad, i'll add a switch in the machine limits tab.

Thank you 

>     2. understood, i'll make a pass on these, i didn't do a good enough job when i added these two boxes.

maybe 2 is for 3 thanks for this
 
> to add images, just drop them on the writing panel from your file explorer.

So simply? Sorry, maybe next time i will try

Side note: when a new version is planned? (I'm using the Linux version) 

Thanks and Regards

Carlo D.

supermerill, 2020-06-01:
When most bugs will be corrected.

commit f2a503cd (2020-06-01): fix #260: optional M20X header
closed 2020-06-01 by supermerill