supermerill/SuperSlicer #261: Outer perimeters alternating direction - can this be disabled?

AndrewEllis93, 2020-05-31:
### Version
2.2.50

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 / SKR 1.4 / Klipper
Voron 2.2 / SKR 1.3 / Klipper

### Behavior
The outer perimeters of my prints seem to alternate direction every layer. I can't figure out how to disable this, if such an option exists. It causes my prints to have a rough surface finish, direction seems to affect corner blobbing etc. 
See attached image. Gold test cube was done with Prusa Slicer, green test cube was done with SuperSlicer. 
[image]

### Is this a new feature request?
If this setting doesn't exist, then it might be!

[3mf.zip](https://github.com/supermerill/SuperSlicer/files/4708201/3mf.zip)

supermerill, 2020-05-31:
You can uncheck "more perimeters".

duplicate of https://github.com/supermerill/SuperSlicer/issues/206

fixed in the next version, pick a nightly to have the fix right now.

AndrewEllis93, 2020-05-31:
Excellent, thank you!

closed 2020-05-31 by AndrewEllis93