supermerill/SuperSlicer #262: Export Config - Value out of range : bridge_overlap

vince2bir, 2020-06-03:
### Version
2.2.49

### Operating system type + version
WIN 10

### 3D printer brand / version + firmware version (if known)
SOVOL S01

### Behavior
I wanted to update to latest release (first update) and I was looking for a way to keep my settings. (I did not find a "how to upgrade" btw).
Found out it could probably be done with the "Export Config" or Bundle.
When I try this, I have an error "Value out of range : bridge_overlap"

I've checked the value of Bridge overlap in user presets in "Printer settings", they are all at 100%.
(I have no system preset (could that be the problem?)).

supermerill, 2020-06-03:
fixed in new version please update.

vince2bir, 2020-06-03:
Thanks. Sorry for the duplicate entry.
How can I transfer my settings to new version then ? :)

neophyl, 2020-06-03:
Config files are stored in the same place for super slicer so running should pick up all your configurations automatically. If they don’t or you want to keep the same settings between Prusa Slicer and Super Slicer then the export config bundle/import config bundle is the way to go. 

On Windows You don’t actually need to install, if you extract the files you can run the exe and it will just run.  That way you can start any version you have extracted. Just don’t run them at the same time as they all access the same configuration folder.   Not sure about versions on other operating systems.

vince2bir, 2020-06-03:
As there was a name change, config files need to be moved as well.

I found the original files here (change [username] with your windows account name :
`C:\Users\[username]\AppData\Roaming\Slic3r++\`
and move them to the :
`C:\Users\[username]\AppData\Roaming\SuperSlicer`

supermerill, 2020-06-03:
oh, yes, i forgot that, I have to write some lines on that int he release page

closed 2021-04-27 by supermerill