supermerill/SuperSlicer #263: Flow Calibration Model Not Correct

sg000, 2020-06-03:
### Version
_Version of SupserSlicer used goes here_
2.2.51

_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Win10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Based on CR10S

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
Running the Temperature Calibration for default 2 up and 2 down, got 4 towers, 5 plaques for the temps and a giant tower looking thing.

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_
[image]

[smart_compact_temp_calibration.zip](https://github.com/supermerill/SuperSlicer/files/4726628/smart_compact_temp_calibration.zip)

sg000, 2020-06-03:
Just started using this slicer btw, LOVE IT!. I was using Cura for months, then last couple weeks with PS was a dream to work with, and this tool looks like the small tweaks I needed with PS!

supermerill, 2020-06-03:
there is a problem with the nozzle scaling.
it works with 0.4 nozzle but i messed a thing when it's not. it should have only grow by 50% in xy....

sg000, 2020-06-04:
> 
> 
> there is a problem with the nozzle scaling.
> it works with 0.4 nozzle but i messed a thing when it's not. it should have only grow by 50% in xy....

Sorry for the bad title, I was working on finishing my flow test when I wrote this. :)
Armed with that info, I changed the nozzle size to .4, then generated the test, then changed it back to .6 and re-sliced and looks like everything is ok.

supermerill, 2020-06-05:
should be corrected in the 2.2.51.1

kvk15, 2020-06-07:
Hello.
The source code on the release page is not updated and remains 2.2.51.
https://github.com/supermerill/SuperSlicer/archive/2.2.51.1.tar.gz - 404 not found.

supermerill, 2020-06-07:
if you want the source code, use git reset --hard c68e94daf2339888ae8f9e5b196075ea00bd5af0

closed 2020-06-24 by supermerill