supermerill/SuperSlicer #264: Error Compiling in Ubuntu

lavachemist, 2020-06-04:
### Version
_Version of SupserSlicer used goes here_2.2.51

_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ Ubuntu 18.04.4
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_ 

While compiling from source in Ubuntu, I get the following errors:

In file included from /home/laser/SuperSlicer/src/test/test_data.cpp:1:0:
/home/laser/SuperSlicer/src/test/test_data.hpp: In function ‘bool Slic3r::Test::_equiv(const T&, const T&)’:
/home/laser/SuperSlicer/src/test/test_data.hpp:61:77: error: ‘epsilon’ is not a member of ‘Slic3r::Geometry’
 bool _equiv(const T& a, const T& b) { return abs(a - b) < Slic3r::Geometry::epsilon; }
                                                                             ^~~~~~~
/home/laser/SuperSlicer/src/test/test_data.cpp: In function ‘void Slic3r::Test::init_print(Slic3r::Print&, std::initializer_list<Slic3r::Test::TestMesh>, Slic3r::Model&, Slic3r::DynamicPrintConfig*, bool)’:
/home/laser/SuperSlicer/src/test/test_data.cpp:268:76: error: cannot bind non-const lvalue reference of type ‘Slic3r::DynamicPrintConfig&’ to an rvalue of type ‘Slic3r::DynamicPrintConfig’
  DynamicPrintConfig &config = Slic3r::DynamicPrintConfig::full_print_config();
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/laser/SuperSlicer/src/test/test_data.cpp:288:63: error: no matching function for call to ‘Slic3r::Model::arrange_objects(double)’
     model.arrange_objects(print.config().min_object_distance());
                                                               ^
In file included from /home/laser/SuperSlicer/src/test/test_data.hpp:6:0,
                 from /home/laser/SuperSlicer/src/test/test_data.cpp:1:
/home/laser/SuperSlicer/src/test/../libslic3r/Model.hpp:805:14: note: candidate: bool Slic3r::Model::arrange_objects(const Slic3r::Print&, const Slic3r::BoundingBoxf*)
     bool     arrange_objects(const Print& print, const BoundingBoxf* bb = NULL);
              ^~~~~~~~~~~~~~~
/home/laser/SuperSlicer/src/test/../libslic3r/Model.hpp:805:14: note:   no known conversion for argument 1 from ‘double’ to ‘const Slic3r::Print&’
/home/laser/SuperSlicer/src/test/test_data.cpp: In function ‘void Slic3r::Test::init_print(Slic3r::Print&, std::initializer_list<Slic3r::TriangleMesh>, Slic3r::Model&, Slic3r::DynamicPrintConfig*, bool)’:
/home/laser/SuperSlicer/src/test/test_data.cpp:301:76: error: cannot bind non-const lvalue reference of type ‘Slic3r::DynamicPrintConfig&’ to an rvalue of type ‘Slic3r::DynamicPrintConfig’
  DynamicPrintConfig &config = Slic3r::DynamicPrintConfig::full_print_config();
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/laser/SuperSlicer/src/test/test_data.cpp:320:63: error: no matching function for call to ‘Slic3r::Model::arrange_objects(double)’
     model.arrange_objects(print.config().min_object_distance());
                                                               ^
* _Steps needed to reproduce the problem_

This happens when I compile the source code. I have edited the g-code writer (I'm working on the Formlabs Form1+ integration we discussed previously), but the errors seem to be unrelated and are happening in the tests folder.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

supermerill, 2020-06-04:
> In file included from /home/laser/SuperSlicer/src/test/test_data.cpp:1:0:

src/test doesn't exist anymore
i suspect that you pull & make, but you have to redo the cmake before.

lavachemist, 2020-06-05:
I cloned the repository again, and was able to get past this particular issue, but I still ran into issues with the other tests. I haven't figured out how to compile without the test issues, so I actually commented out large sections of the test just to make SuperSlicer compile.

[image]

supermerill, 2020-06-05:
I don't build tests on linux. So there may be incompatibilities between compilers, hence the errors.

I don't know how you manage to make cmake to build them, btw.

supermerill, 2020-07-03:
it's okay now?

closed 2020-07-03 by supermerill