supermerill/SuperSlicer #266: Different resources folder between master and release

pix, 2020-06-04:
### Version
2.2.51 release and tags 2.2.51 on git

### Operating system type + version

Ubuntu 20.04 (also on 19.10)

####
cmake command: `cmake ../ -DCMAKE_BUILD_TYPE=Debug -DSLIC3R_WX_STABLE=1 -DSLIC3R_FHS=0 -DCMAKE_PREFIX_PATH=/home/pix/.superslicer/ -DCMAKE_INSTALL_PREFIX=/home/pix/.superslicer/ -DSLIC3R_BUILD_TESTS=OFF -DLDFLAGS=-lpthread -DSLIC3R_GTK=3`

### Behavior
When installing (using `make install`) I run into a segfault (After loading print.ul)

```
01:20:22 PM: Debug: ClientToScreen cannot work when toplevel window is not shown
[New Thread 0x7fffdd737700 (LWP 254955)]
[New Thread 0x7fffdcf36700 (LWP 254956)]
create settings  print.ui
terminate called without an active exception

Thread 1 "superslicer" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff638e859 in __GI_abort () at abort.c:79
#2  0x00007ffff7e4d951 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff7e5947c in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff7e594e7 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x0000555555c81483 in std::thread::~thread() (this=0x555558d31c10, __in_chrg=<optimized out>) at /usr/include/c++/9/thread:139
#6  0x00005555562bc1ba in Slic3r::GUI::Mouse3DController::~Mouse3DController() (this=0x555558d31a90, __in_chrg=<optimized out>)
```

It turns out I works fine when I rsync the `resources` folder from the 2.2.51 release (They are indeed different)

pix, 2020-06-04:
Strange...
[image]

pix, 2020-06-04:
Even more so...
[image]

pix, 2020-06-04:
Something about line endings seems fishy:

```
diff -rub resources SuperSlicer/resources | wc -l
0
```

pix, 2020-06-04:
It's caused by git's `core.autocrlf = true`.
It seems the parser is expecting \r\n line endings in ui files.

pix, 2020-06-04:
A kind of solution is discussed here: https://stackoverflow.com/questions/6089231/getting-std-ifstream-to-handle-lf-cr-and-crlf/6089413#6089413

This seems to do the trick: 

```diff
diff --git i/src/slic3r/GUI/Tab.cpp w/src/slic3r/GUI/Tab.cpp
index 19d427028..b71e2bf96 100644
--- i/src/slic3r/GUI/Tab.cpp
+++ w/src/slic3r/GUI/Tab.cpp
@@ -39,6 +39,38 @@
 namespace Slic3r {
 namespace GUI {
 
+std::istream& safeGetline(std::istream& is, std::string& t)
+{
+    t.clear();
+
+    // The characters in the stream are read one-by-one using a std::streambuf.
+    // That is faster than reading them one-by-one using the std::istream.
+    // Code that uses streambuf this way must be guarded by a sentry object.
+    // The sentry object performs various tasks,
+    // such as thread synchronization and updating the stream state.
+
+    std::istream::sentry se(is, true);
+    std::streambuf* sb = is.rdbuf();
+
+    for(;;) {
+        int c = sb->sbumpc();
+        switch (c) {
+        case '\n':
+            return is;
+        case '\r':
+            if(sb->sgetc() == '\n')
+                sb->sbumpc();
+            return is;
+        case std::streambuf::traits_type::eof():
+            // Also handle the case when the last line has no line ending
+            if(t.empty())
+                is.setstate(std::ios::eofbit);
+            return is;
+        default:
+            t += (char)c;
+        }
+    }
+}
 
 wxDEFINE_EVENT(EVT_TAB_VALUE_CHANGED, wxCommandEvent);
 wxDEFINE_EVENT(EVT_TAB_PRESETS_CHANGED, SimpleEvent);
@@ -1151,7 +1183,7 @@ bool Tab::create_pages(std::string setting_type_name, int idx_page)
     //read file
     std::ifstream filestream(ui_layout_file.c_str());
     std::string full_line;
-    while (std::getline(filestream, full_line)) {
+    while (safeGetline(filestream, full_line)) {
         if (full_line.size() < 4 || full_line[0] == '#') continue;
         //remove spaces
         while (full_line.size() > 1 && (full_line.front() == ' ' || full_line.front() == '\t')) full_line = full_line.substr(1, full_line.size() - 1);
```

supermerill, 2020-06-04:
i'm using core.autocrlf=true with git, so each os should have their own line ending.

Did you clone on windows then copy the directory into your ubuntu?

pix, 2020-06-05:
No, as you can see it on this screenshot: https://github.com/supermerill/SuperSlicer/issues/266#issuecomment-638994418

supermerill, 2020-06-05:
well, i'll merge your patch, better be safe.

pix, 2020-06-05:
I Have a patch using boost's newline filter if you want in the next minutes :)

pix, 2020-06-05:
[0001-266-makes-sure-we-re-normalizing-line-endings.txt](https://github.com/supermerill/SuperSlicer/files/4736865/0001-266-makes-sure-we-re-normalizing-line-endings.txt)

supermerill, 2020-06-05:
problem is, if you ave problems with these ui files, you will have the same problem with the profile ones.

supermerill, 2020-06-05:
I just re-checked all files in the release, and they all have their own good cr / lf.

also, at least in windows, it doesn't matter if you have cr lf or only lf, it work as-is.

i don't know why your compiler on linux want a windows-style line ending. Or maybe i'm not following you.

pix, 2020-06-05:
It doesn't happen inside `PresetBundle::load_config_file` because it's using boost_read_ini which is calling [trim on every lines](https://github.com/mousebird/boost/blob/591b85a17ba2e2075804d7ffdb85212ba49ea23f/boost/property_tree/ini_parser.hpp#L93)

```bash
git -c core.autocrlf=false clone --depth 1 https://github.com/supermerill/SuperSlicer.git SuperSlicer.git.false
git -c core.autocrlf=true clone --depth 1 https://github.com/supermerill/SuperSlicer.git SuperSlicer.git.true
file SuperSlicer.git.*/resources/ui_layout/Readme.md 
SuperSlicer.git.false/resources/ui_layout/Readme.md: ASCII text
SuperSlicer.git.true/resources/ui_layout/Readme.md:  ASCII text, with CRLF line terminators
```
Maybe release works because `core.autocrlf=false` on automated builds ?

pix, 2020-06-05:
This seems to achieve the same purpose (it works with both \n and \r\n line endings):

While making the code clearer, what do you think ?

```diff
diff --git i/src/slic3r/GUI/Tab.cpp w/src/slic3r/GUI/Tab.cpp
index ec8ca606b..18790907b 100644
--- i/src/slic3r/GUI/Tab.cpp
+++ w/src/slic3r/GUI/Tab.cpp
@@ -23,6 +23,7 @@
 #include <wx/settings.h>
 #include <wx/filedlg.h>
 
+#include <boost/algorithm/string/trim.hpp>
 #include <boost/algorithm/string/split.hpp>
 #include <boost/algorithm/string/predicate.hpp>
 #include <boost/filesystem.hpp>
@@ -1159,10 +1160,8 @@ bool Tab::create_pages(std::string setting_type_name, int idx_page)
     std::ifstream filestream(ui_layout_file.c_str());
     std::string full_line;
     while (std::getline(filestream, full_line)) {
-        if (full_line.size() < 4 || full_line[0] == '#') continue;
         //remove spaces
-        while (full_line.size() > 1 && (full_line.front() == ' ' || full_line.front() == '\t')) full_line = full_line.substr(1, full_line.size() - 1);
-        while (full_line.size() > 1 && (full_line.back() == ' ' || full_line.back() == '\t')) full_line = full_line.substr(0, full_line.size() - 1);
+        boost::algorithm::trim(full_line);
         if (full_line.size() < 4 || full_line[0] == '#') continue;
         //get main command
         if (boost::starts_with(full_line, "logs"))
```

supermerill, 2020-06-05:
core.autocrlf = true should pull with your os line ending.

Your last patch seems good, i'll try and push it if all green.

pix, 2020-06-05:
I got a failed build using the last version

`no_page_yet` isn't always defined but it's unconditionnaly returned by `Slic3r::GUI::Tab::create_pages`

Otherwise, it works !

supermerill, 2020-06-05:
oh, yes, you're right, i have to remove it from the win-only section.

closed 2020-08-24 by supermerill