supermerill/SuperSlicer #272: Bridge Flow Calibration Issue

paulhew69, 2020-06-06:
### Version
2.2.51

_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Ubuntu

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Error after clicking 'Test Flow Ratio' button in Bridge Flow Calibration.
[image]
Workaround, Go to the 3D Editor View, Autoarrange, then click Slice now.

supermerill, 2020-06-06:
Can you post the project file when the problem appear? thx

paulhew69, 2020-06-06:
It was using the Bridge Calibration tool in SuperSlicer, so it is part of your suite of tools, not one of my files.

Regards

supermerill, 2020-06-07:
yes, but once you clic on it, you will have the error pop-up and then you can save and send it to me.
It's not only for the generated geometry but also for your printer settings. 
The calibration should generate something for your printer, if it's failing and I don't have your printer conf, i can't replicate nor correct the issue.

paulhew69, 2020-06-07:
[image]
Sorry, now I see what you are referring too.
I have a 300 x 300 bed.

If you click back to the '3d editor view' you see this.
[image]

If you then click Auto-Arrange it puts things where they should be.
[image]

Then click 'Slice Now' It is able to complete correctly
[image]

Hope this info helps

thierryzoller, 2020-06-07:
Have had the same issue, you need to delete a few objects and push them further apart, then it works.

supermerill, 2020-06-07:
please use File -> save as
then zip the saved .3mf file
then drop it here in a comment.

paulhew69, 2020-06-07:
As requested

Regards,
Paul
[bridge_test with issue.zip](https://github.com/supermerill/SuperSlicer/files/4742092/bridge_test.with.issue.zip)

closed 2020-08-24 by supermerill