supermerill/SuperSlicer #273: French language  ?

le-belge, 2020-06-06:
### Version
_Version of SupserSlicer 2.2.51_

OS : win 10 64 bit 
Hi ,
great job on this software!
I love...
is it possible to have it in French?
Thanks
Mike

supermerill, 2020-06-06:
i'm doing a pass on the french traduction file sometimes.
Currently, i think that i forgot to renamed all the traduction files, so there is nothing instead of half.
I'll do something for it for the next release.

le-belge, 2020-06-06:
excellent, that's great, thank you!

neophrema, 2020-06-09:
I'm sorry, a little bit off-topic.
I found some spelling mistakes in Superslicer, should I file a report and where? Is it appreciated, since its just "cosmetics"?
E.g.:  Perimeters&shell -> Quality -> extra perimeters on overhang - Tooltip. 
check the writing please.

supermerill, 2020-06-09:
ideally, you create a fork with the button int he top, then you modify the file (directly in github, search for it, if it's a setting, it's in printconfig.cpp.
then apply/commit, then push the "pull request" that should  pop (in the root of your repo).
That will create a request for me to accept your changes.

neophrema, 2020-06-09:
unfortunately I would have to look into a lot of (rewarding) details, since I do not use github, nor do I frequently compile programms myself (I'm not into programming). I simply don't have time for this right now, but maybe in a Month (a lot of duties right now). Still, I would like to help making something better, that's worth it (I appreciate your work). If you wouldn't mind I would come up with a list of spelling mistakes in the programm and list it? For this I think I should make a new issue;
Would that be helpful? Otherwise I can do like you requested but not now, but in a Month, since I have too much to do with University and Work.
I'm sorry. -
Which way to go?

supermerill, 2020-06-09:
You don't need to install anything.

search the text you want o change in the search field.
Click on the file, then on the edit button (it's a pencil)
modify the text.
then click on propose change.

(but it's better if you modify all the bad string in the file before clicking on propose

see tuto here: https://help.github.com/en/github/managing-files-in-a-repository/editing-files-in-your-repository

If you give me a list of "replace that by this", that's exactly what i'm going to do anyway.
If you really can't use the search box and clicking on the edit button, then post the needed changes here, i will do the work.

neophrema, 2020-06-09:
Thank you very much for your explanation. I will do so till the end of the Weekend. See you and thank you!

neophrema, 2020-06-14:
I did a few changes, I will do more, have no time yet.
I hope this is helpful; 

One last request: I think you should number the calibrations due to your advice to use them in ORDER.
Meaningly name them:

1. Bed leveling
2. Flow calibration 
3. Filament ...

and so on.
ALSO: the first few lines in each calibration is regarding which you should have done first.
Therefore I think it should be standardized:

"Before this Calibration you should have done:
- Bed Leveling
- Flow calibration
- Filament ... "

What do you think?

supermerill, 2020-06-14:
It's a sensible improvement.
Maybe also I should add a How-to page with no calibration, but a quick explanation on how to use the menu, or what calibration to look after if there is a problem.

neophrema, 2020-06-14:
Yes, I agree. It's a good idea.
I will continue looking for typos and co, there were some outside of the new calibration part; I wasn't able to find the sourcetext, I had to click on a lot of files to find that its in the HTML for some reason.
I hope I can be helpful, thank you very much for your work!

supermerill, 2020-06-15:
oh, sorry, i didn't think of them.
It's done that way because it's easier to modify & format a big chunk of text with images in html.

settings text are in src/libslic3r/PrintConfig.cpp
some ui text (like the tab names) and some settings names are in resources/ui/??.ui
some other bits of settings taht haven't made their way into ui files are in src/slic3r/GUI/Tab.cpp
configuration text are in resources/calibration/??/??.html

le-belge, 2020-06-15:
Hi, this is working hard as far as I can see!

neophrema, 2020-06-15:
Im sorry, what are you referring to le-belge?

supermerill, 2020-06-15:
I think he's just cheering us up.

neophrema, 2020-06-15:
I just understood, yes i guess so- still, maybe he did misunderstood his personal issue since I was hijacking this thread. (I think you answered him before so I guess its ok.- I will look after the files you mentioned in the next few days; lot to do right now)
sincerely,
v.

le-belge, 2020-06-15:
no problem, I just wanted to say that it was good to see you work, I hope to have soon french to fully exploit this superb software!

closed 2020-10-01 by supermerill