supermerill/SuperSlicer #274: src/slic3r/CMakeLists.txt needs updating

MakerMatrix, 2020-06-07:
In the latest code, as of tonight, cmake fails because you have two FreeCad GUI files listed in CMakeLists.txt that don't exist in the src/slic3r/GUI directory:

---------
CMake Error at src/slic3r/CMakeLists.txt:206 (add_library):
  Cannot find source file:

    GUI/FreeCADDialog.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error in src/slic3r/CMakeLists.txt:
  Cannot find source file:

    GUI/FreeCADDialog.hpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
------------

MakerMatrix, 2020-06-07:
On further inspection it appears that perhaps those two files were not added to the repository before the commit earlier today.  I can't get it to build without them.

supermerill, 2020-06-07:
i forgot to add them to the commit.
And i just moved to an other house for the week-end....
i'm sad, can't work on them for some days.

closed 2020-06-17 by supermerill