supermerill/SuperSlicer #275: BUG - Usage of "below" instead of above?

thierryzoller, 2020-06-07:
This confused me : )
Printer Settings ->  Machine limits - Usage Tab

It says : "If the setting below it checked" , shouldn't it be "if the setting **above** is checked" ?

thierryzoller, 2020-06-07:
I just did some tests and yes it should say "above".

Also I would suggest clarifying "This tab" as I understood it as only the "Usage section".

supermerill, 2020-06-07:
Do you have a suggestion on how i can rephrase it?

thierryzoller, 2020-06-08:
I would completely relable it for clarity:

Title:  Machine Limits
Checkbox: Enable Limits
Description: The information below information is only used to calculate estimated printing time, unless you actually enable the option.

supermerill, 2020-06-08:
Done.
I wrote "above" because i added the comment before the checkbox, it's strange that they show in the opposite order, i may have forgot to call something...

closed 2020-08-24 by supermerill