supermerill/SuperSlicer #276: Single nozzle/Dual materials+Klipper Gcode flavor issues

Alonso3D, 2020-06-07:
SuperSlicer_2.2.51

Windows 10 64bit

Voron 2.2 running Klipper on Dual SKR 1.3 boards with RPi 3b+ and Dual extruders with single nozzle

1. Slicing preview fails to display colors for each extruder when set up for single nozzle with multimaterials (2ea in this case)

This seems to be correlated with the Gcode flavor as this problem does not occur when using Marlin Gcode flavor.
Using Marlin gcode flavor displays the colors for each extruder correctly.

2. The print errors out and cancels due to the gcode calling "ACTIVATE_EXTRUDER EXTRUDER=extruder0" in lieu of "ACTIVATE_EXTRUDER EXTRUDER=extruder" with klipper gcode flavor selected.
Perhaps it should be users choice what to name the extruder/tool within SuperSlicer or have it use standard tool change Gcode, such as T0 and T1, etc. Thus the user would need T0/T1 tool activation macros in their printer cfg file in order to use standard Tool switching gcode.

One should notice that klipper configs never have "0" in the name for when a single is component is used. Numbers are used when multiple components are used such as steppers or heaters. 
e.g. - Stepper_z, Stepper_z1, Stepper_z2 and so 

[image]
[image]

[Projects.zip](https://github.com/supermerill/SuperSlicer/files/4742780/Projects.zip)

supermerill, 2020-06-07:
right now, extruders doesn't have name. SO there is multiple possibilities:
 - add a name field, that can be a number (for marlin) or a string (for klipper)
or
 - detect when the right g-code is added in the toochange custom g-code and use this one instead of the auto-generated one.

in the meantime, i can remove the '0' for klipper.

Alonso3D, 2020-06-07:
Extra name field for Marlin isn't necessary as T code for selecting tools is standard.

Klipper is the special case as it can go either way by using a standard command to activate extruder,  extruder1, extruder2 etc or it can use T tool change codes as long as the user has macros to support the T codes for tool changing. Perhaps have a check box when selecting klipper gcode flavor to have the option to use T codes or not(beginning with T0 like marlin)

closed 2020-08-24 by supermerill

kackle1998, 2021-05-27:
Hello, I am also experiencing this issue, where I need to customize my T0, T1, T2...etc... otherwise will run into a crash.

supermerill, 2021-05-28:
> Hello, I am also experiencing this issue, where I need to customize my T0, T1, T2...etc... otherwise will run into a crash.

Can you create a new issue, with project file...etc... otherwise I can't understand your problem.

kackle1998, 2021-05-28:
> > Hello, I am also experiencing this issue, where I need to customize my T0, T1, T2...etc... otherwise will run into a crash.
> 
> Can you create a new issue, with project file...etc... otherwise I can't understand your problem.

Sure, I apologize, I'm new to Git. I'll make an issue.