supermerill/SuperSlicer #278: Minimum extrusion length

guestisp, 2020-06-09:
is possible to add a setting to prevent very very short extrusions that most of the time would lead to blobs and artefacts ?

In example here:

[image]

The "point" inside the circle is very very very short. With bowden setups and PETG, most of the time this is overextruded due to oozing and similiar.

Is possible to skip this , by setting a min length (in mm) somewhere?

supermerill, 2020-06-09:
did you tried to increase gapfill -> min surface ?

guestisp, 2020-06-09:
No, because it isn't a gap-fill, it isn't marked in white in the preview.

supermerill, 2020-06-09:
oh, right. I'll make something to transform it to gapfill (that can be better printed or be eliminated).

guestisp, 2020-06-09:
That would be perfect

supermerill, 2020-08-17:
i've added a min length feature, but it's more to avoid congestion in serial communication.
can i have your .3mf to debug this?

StrikeEagleCC, 2021-03-16:
I think the default value for this is a bit too conservative, and/or something changed between 2.3.52 and 2.3.55. In 2.3.55, a 50mm "arc" with minimum extrusion length set to the default of 0.035 results in 2mm line segments, and visible vertices on the arc surface. However, I like that the setting is there, so thank you.

supermerill, 2021-04-25:
@StrikeEagleCC  I found recently a bug in the min_length feature, so now it should behave more sanely.

JonBerlin, 2021-11-30:
Where can I find the **min_length setting**?

neophyl, 2021-11-30:
Its under Printer Settings>General>Firmware>Processing Limit.

closed 2022-07-14 by supermerill