supermerill/SuperSlicer #280: Allow arbitrary infill rotation for extra toughness

Sidic101, 2020-06-11:
### Behavior
Apparently there are some fibrous structures in the "boxing gloves" of mantis shrimps that strongly resemble solid rectilinear infill.

https://www.tested.com/science/496090-biomimetics-studying-striking-power-mantis-shrimp/

The difference is, each layer is rotated approximately 15°, as opposed to the 90° typically used for 3d prints. It may be worth a try to experiment with different rotation angles to try to match the mantis shrip's increfible toughness and impact resistance. Some people seem to imply that the helical layer arangement makes the structure signifficantly mode difficult to shatter.

liftbag, 2020-06-11:
Do you mean something like this? This is one degree infill rotation per layer, available on the death project Pathio slicer.
https://www.youtube.com/watch?v=PtDG26JRJxY&feature=youtu.be

Sidic101, 2020-06-11:
Exactly this, but I'd like for this to be available for rectilinear infill and to be able to set the angle shift per layer

liftbag, 2020-06-11:
> Exactly this, but I'd like for this to be available for rectilinear infill and to be able to set the angle shift per layer

Yes. For the video I chose a double 90 ° angled texture with 1 degree of inclination per layer, but it is also possible to choose the single texture as in the following videos.

1° per layer
https://youtu.be/pkOjIuaWJuw

15° per layer
https://youtu.be/phW77iHglMQ

I vote for this feature request too.

supermerill, 2020-06-11:
To summarize, currently, some infill use a "90° rotate per layer" (like rectilinear) and this issue is about creating a setting that allow to set this "infill turn per layer" ? 

Or do you want it to be applied to every infill (even ones that don't rotate like cubic and gyroid) ?

liftbag, 2020-06-11:
> To summarize, currently, some infill use a "90° rotate per layer" (like rectilinear) and this issue is about creating a setting that allow to set this "infill turn per layer" ?
> 
> Or do you want it to be applied to every infill (even ones that don't rotate like cubic and gyroid) ?
> 
> Just a note: infill doesn't give toughness, or marginally. And I don't think any new things can increase it to warrant the work. Can be useful for artistic stuff, though.

I think it should be limited to rectilinear, grid and triangles, as in the following video. Just my two cents.

Pathio rorative infill: 120° (triangle) rotated by 15° each layer.
https://youtu.be/vT_cjPSyV10

Sidic101, 2020-06-11:
> To summarize, currently, some infill use a "90° rotate per layer" (like rectilinear) and this issue is about creating a setting that allow to set this "infill turn per layer" ?
>
> Or do you want it to [...]

I don't think this would really make sense for 3d infills like gyroid or cubes, but grid and triangles may get some cool looking results.

The use case I originally had in mind was 100% rectillinear infill on parts that need maximum compressive strength and a lot of toughness (e.g. a vice spacer, ablative plastic bearings, replacement machine parts).

Sidic101, 2020-06-11:
feel free to correct me if infill won't help with those cases either

neophyl, 2020-06-11:
I think Cura let’s you specify 2 numbers, so the initial angle and then a step rotation value too for some infill.

supermerill, 2020-06-11:
> feel free to correct me if infill won't help with those cases either

i mean, at 100%, it's 100%, perimeter of infill.
I was talking at more infill vs more shell, at constant weight.

I'll look into it, seems easy, but the last time said that...

closed 2020-10-22 by supermerill