supermerill/SuperSlicer #281: Speed Adjustment on Plater sheet

Cameluntu, 2020-06-11:
### Version
2.2.50

### Operating system type + version
windows_

### 3D printer brand / version + firmware version (if known)
BLV / Creality Ender3 Marlin 2.0

### Behavior
* _Feature Request_

Hi, i am just throwing out there a new feature request of a field i am missing out from other slicers that i have been using. 
Its not vital but i believe its a nice simple add-on.

Depending on the work i am about to print, i need to change overall speeds.
Since i have a fixed ratio between differents speeds, it would be extremely helpful and easy to have a slider or dropdown menu where a user can decide at what percentage shall the machine print.
Speeds than will be internally scaled to the selected percentage.
Slider/Dropdown box can be placed next to the infill dropdown list in the plater sheet.

Keep up the awesome work!
Andrea

neophyl, 2020-06-12:
Don't you have a speed setting on the actual printer firmware ?  Both my Prusa and my Cr10 I can adjust the overall print speed using the control on the front to speed up or slow down.

Cameluntu, 2020-06-12:
Yes, I do however I think it is preferable to do it from the slicer since a user can analyze how changes affect its print instead of just throwing a random number based on feelings from a printer screen. Another thing is, for those who uses octoprint and such with the direct print option from superslicer it's a quicker way to adjust speeds. Again, if there is no interest in this (users base) I guess we can close this request

supermerill, 2020-06-12:
One advantage vs the firmware thing, is that is doesn't increase the travel speed, that may already be the fastest speed possible.

Cameluntu, 2020-06-13:
that sounds a nice solution indeed! thanks a lot for taking this into account!

wesc23, 2020-06-13:
Check Kiss's implementation of this.  You give a range for various line types and then have a slider from slow to fast to interpolate.  Works really well and is WAY less fiddly than adjusting a dozen different speeds in the speed tab.