supermerill/SuperSlicer #282: Feature Request: Pressure Advance Tuning

SnowDrifterr, 2020-06-12:
Would be cool, if possible, to have some way of tuning or setting up a script to easily tune pressure advance. Marlin users have an easy code generator available to them. Duet users do not

mdealer, 2020-06-12:
Isn't this best accomplished by Duet Devs themselves?

supermerill, 2020-06-12:
> Would be cool, if possible, to have some way of [tuning or] setting up a script [...]

You can enter your script in output-> post-processing scripts.
That's what you asked for?

digitalentropy, 2020-10-09:
I think this is a great idea. Having to use the Marlin website and re-enter printer and filament settings from scratch is not ideal. Being able to generate LA or PA tests from within SS would be an *excellent* addition.

supermerill, 2020-10-09:
I have to research for each stl & procedure for each firmware (klipper, marlin, reprap...)
if you know where they are, put the links here.

klipper: https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md

marlin: generate a special gcode froma  generatr (https://marlinfw.org/tools/lin_advance/k-factor.html) 

reprap: use the marlin one...

maybe i can use the klipper one for every printer, just with varying gcode for pa setting.

KrazzyTOSSER, 2020-10-29:
Simple and convenient script to generate a test model for Presure Advance: http://www.sternwolken.de/tmpup/3dprint/pressureadvance2.py
Description of the script:
https://forum.duet3d.com/topic/6698/pressure-advance-calibration/134?_=1603999017043

supermerill, 2020-10-29:
Are you the one who makes it?
I can't use it unless the dev release it under agplv3.

KrazzyTOSSER, 2020-10-29:
Sorry, no, I'm not the author. Didn't think about the license.

supermerill, 2020-10-29:
maybe we can ask, do you know how to send a message to him? on the duet forum? i didn't found the the link on the linked page.

KrazzyTOSSER, 2020-10-30:
To write a private message to the user, you must register on the forum. After that, you can go to author profile:
https://forum.duet3d.com/user/corezair
then click on the symbol of three dots and choose to start a chat.

KrazzyTOSSER, 2020-10-30:
Also, this script is suitable for testing Linear Advance if marlin is used. You only need to replace all M572 with M900