supermerill/SuperSlicer #286: Feature Request:  Smart Lift Z only on layers above overhangs

wesc23, 2020-06-12:
### Version
2.2.51

### Operating system type + version
MacOS Sierra

### 3D printer brand / version + firmware version (if known)
N/A
### Behavior

Z lift is most useful to avoid snagging the nozzle on an upcurled edge from overhangs on previous layers.    It would be great if SS had a feature where you could specify "Only Z Lift N layers above an overhang".  This would save a lot of print time as well as wear on leadscrews.

The feature would only activate if N is > 0.   Another option would be to have a checkbox plus a numeric entry field.

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs

N/A

supermerill, 2020-06-12:
It's difficult, you have to create maps of where there are overhangs, try to avoid these when travelling, or lift if you can't avoid.

Why not, but i can't do it in the coming weeks/months.

wesc23, 2020-06-12:
Yeah, was asking for the simple version - just a flag indicating an overhang in that layer.  A heatmap version would be nice, but as you say, that's difficult.