supermerill/SuperSlicer #287: Feature Request: xy compensation in %

ufoDziner, 2020-06-13:
### Version
2.2.51

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Voron 2.2 and Bear MK3

### Behavior
FEATURE REQUEST
I use shrinkage percent compensation settings in my current slicer to adjust for per-material shrinkage in ABS, or for HTPLA+ parts that will be annealed.  It would be great to have the xy compensation settings have a percentage option along with the current mm option.

supermerill, 2020-06-13:
How the shrinkage affect holes?
if you extrude a tube, will the walls contract and the hole grow or will the hole stay the same size (or shrink?) because the part as a whole contract at the same time?

If we want to add a "filament shrinkage compensation", we must do it right.
I don't have much time to do experiments with big part, small holes everywhere, big hole, etc. But i or you have to do that to understand what i have to add to be able to correct it the best way possible.

ufoDziner, 2020-06-13:
It is my experience that holes get smaller at the same rate.  Meaning, the entire part gets smaller.  Using the part scaling on the main screen manually does the proper correction.

If you have a part that has the qualities that you'd like tested, I will run the tests and report the results.

supermerill, 2020-06-13:
if we have a part made with 2 filaments (with a tool-change printer for exemple, or a prusa mmu), and they have very different shrinkage properties... I don't think it's a good idea to change the dimension of the object dynamically... it should  stay the same from start to finish, right?

There is two way that i can think of to resolve this:
 - set this setting in the print / object setting, so it isn't dependent of the filament
 - use the smallest retraction in all filaments used

Also, the first layer(s) shrink the same way as other or do we have to do something for these?

ufoDziner, 2020-06-13:
I'm not sure about the mixed-material parts.  I would think that it would be best to have the end result match otherwise the part would end up under constant stress.  It would certainly be best to get input from someone with mixed-material part experience.

For me, it's really filament dependant.  Only ABS requires compensation, no others.

The first layer appears to be the same as the rest.  I still use additional elephant foot compensation to achieve the desired results.

supermerill, 2020-06-13:
> I still use additional elephant foot compensation to achieve the desired results.

Maybe that's something that can also be added when using this feature, adding a bit less xy growth for the 1-3 first layers (scaled)

supermerill, 2020-08-19:
Next version will have a filament shrinkage setting.

khumarahn, 2020-09-18:
Just tried to use non 100% shrinkage. The slicer crashes on attempt to slice anything, even the built-in shapes like cylinder or box. Tried with the latest appimage 2.2.53.2. Setting shrinkage to 100% makes it work again.

khumarahn, 2020-09-18:
Ah, I made a dumb mistake. I entered shrinkage of 0.995 instead of 99.5. With 99.6 it works fine. Sorry. Though maybe add a check or a warning?

supermerill, 2020-09-19:
I'll change the min to 10%

closed 2020-10-22 by supermerill

manolitto, 2021-01-13:
> Next version will have a filament shrinkage setting.

Why is scaling only done in x and y direction? If we are speaking of CTE compensation here, I guess filament specific auto-scaling in all 3 dimensions would be the way to go, no? Or do I miss something here?

supermerill, 2021-06-02:
>  Why is scaling only done in x and y direction? If we are speaking of CTE compensation here, I guess filament specific auto-scaling in all 3 dimensions would be the way to go, no? Or do I miss something here?

The z shrinkage is compensated by the extrusions of the next layer, as the printer head is always at the right z. You may want to add some flow to compensate if it's really severe and creates under-extrusion.

jasonbcox, 2021-06-14:
I don't follow. If you have a tall print using a filament that shrinks excessively (ie. ABS), you will want to print additional layers to compensate so the final product is the correct dimension along the z-axis.

supermerill, 2021-06-15:
> I don't follow. If you have a tall print using a filament that shrinks excessively (ie. ABS), you will want to print additional layers to compensate so the final product is the correct dimension along the z-axis.
the time you will reach the top, most of the other layers are already cooled.

jasonbcox, 2021-06-15:
So ideally, the printer should compensate for shrinkage along the z-axis as it prints in addition to added layers in the slicer. This primarily impacts tall parts.

supermerill, 2021-06-15:
If you can get data on how the filament shrinkage affects the needed flow (from different filaments types), taking into account the layer duration, I can use it to improve the feature.

ogland, 2021-07-23:
Hi! I have been using this feature to compensate for perimeter shrinking with ABS, but it turns out it scales the whole model in XY, instead of scaling like xy compensation in slicing settings. The tooltip is kinda misleading in that way I think, and I'm not sure if the feature is even useable in the current state as parts doesn't shrink like the function tries to compensate for.

supermerill, 2021-07-25:
> instead of scaling like xy compensation in slicing settings

what does that mean?

ogland, 2021-07-25:
> > instead of scaling like xy compensation in slicing settings
> 
> what does that mean?

Expanding the model outward instead of scaling. ABS shrinkage tends to contract inwards toward material center, more localized than overall scaling.

supermerill, 2021-07-25:
> Expanding the model outward instead of scaling

It's not scaling, as inner perimeters are shorter instead of longer.
And even so, I don't see the difference between "expanding" and "scaling".

> ABS shrinkage tends to contract inwards toward material center, more localized than overall scaling.

I think it's the limit of a generic algorithm that can work with every file possible. "material center" doesn't exist.

ogland, 2021-07-25:
Ok. Maybe I'm explaining it poorly.

Here's a normal sliced benchy:
[image]

Here's with 60% shrinkage defined in filament setting, this would move things a lot for bigger or more complex parts like hole spacing etc:
[image]

Here's what XY size compensation at +5 mm for outer and -5 for inner does, and this (in my opinion) more accurately can counter shrinkage, so this is kind of what I expected the shrinkage function to do:
[image]

All values are of course greatly exaggerated in order to show what's going in.

bryn51, 2024-07-12:
time has passed since this issue was opened, and Prusa Slicer 2.8.0 (and maybe prior) now has Shrinkage compensation for X.Y and Z axis.  I have used it for printing Nylon (Polymaker CoPA (PA6) , but find that holes are coming out smaller (16.7 %) than the model design, whereas the model itself is slightly (0.36 %) larger (this is with  shrinkage compensation adjusted already. So, I am not sure whether this is a bug or what it may be.  Certainly I'd like to see a Kbase article on the topic, or some appearance in the PS User Guide, but I am able to find neither. Is that lack of documentation a bug in and of itself ?