supermerill/SuperSlicer #288: Missing M204 commands with Klipper

shiftingtech, 2020-06-13:
### Version
_Version of SupserSlicer used goes here_
2.2.51
_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Linux/debian testing
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Voron2 with klipper (
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
I have a series of accelerations set in my profile, under Acceleration Control (Advanced).  If I have my printer type set to RepRap, a series of M204 commands are inserted into the gcode, adjusting the acceleration for various feature types. If, On the other hand, I set printer type to "Klipper"  only the very first M204 command appears, and the printer then stays at that acceleration for the entire print.
_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

[voron_design_cube_v6.zip](https://github.com/supermerill/SuperSlicer/files/4774267/voron_design_cube_v6.zip)

supermerill, 2020-06-13:
I though machines limits weren't used in the gcode, but it seems that i was wrong. Sorry for that.

machines limits : maximum acceleration while extruding is at 1500, so it stay at 1500 for all the print. Put a bigger value here.

shiftingtech, 2020-06-13:
So there's probably a new bug there: machine limits not enforced in reprap
mode...

Still, that's awesome, and easy to fix.  Thanks for the quick response!

TheMrCaveman, 2020-06-17:
how to remove machine limits from reprap profile. Don't want this and in prusaslicer this is automaticly disabled when choosing reprap. Mayby make it an option?

supermerill, 2020-06-18:
They don't do anything, they just act as a failsafe in case your print profile isn't in sync with your machine one (reusing a profile with an other printer). Also, they are needed to compute a time estimate.

Prusa hide them, but they are always here.

closed 2020-08-25 by supermerill