supermerill/SuperSlicer #292: Support patterns

espr14, 2020-06-13:
There should be 3 support pattern types:
1. first layer
2. infill
3. model interface

There are only the latter 2 there since the first layer uses the model interface one. In case of ironing it does not make any sense to make the first support layer nice and flat.

Moreover, there could be all patterns from model infill patterns available (triangles, stars, gyroid etc.). I know that some of them do not make much sense but why not leave the decision on the user?

supermerill, 2020-06-13:
> some of them do not make much sense but why not leave the decision on the user?

if it doesn't make sense, it allow for a clearer and shorter GUI.

espr14, 2020-06-13:
Honeycomb does not make much sense but it is there. There are others that are better (strength, print speed).

n8bot, 2020-06-17:
In terms of the first layer of support (forced first layer sparse raft) being the same as the model interface, I agree that this needs to be configurable.

The concentric interface option is truly magnificent. This is something I've wanted for a while but happily stumbled upon in SS a few days ago. However, when we use it, the first layer forced-raft for supports is ludicrous. We need to be able to adjust all parameters for that first layer raft: spacing, pattern, how much it "extends" past the x/y limits of the support.

I personally don't need that raft at all. I wish for the support to just form normally at the first layer. A PEI print bed is great for adhering tiny traces with no need for brims or excessive extrusion.

Any thoughts?

supermerill, 2020-06-17:
i can add some settings in the raft section. From the previous posts:
* pattern (currently rectilinear only ? )
* spacing (currently 50% or 100% only ? )
* "mandatory raft layer under support" (currently mandatory)
* anchor offset

> A PEI print bed is great for adhering tiny traces with no need for brims or excessive extrusion.

I've had problems with infill if there are no solid bottom layer, but my pei isn't great. So it's safer with brim => brim is default

n8bot, 2020-06-17:
That would certainly make me happy!

For your first point, you mention 'currently rectilinear only,' but currently if the interface pattern is concentric, the mandatory raft layer adopts the same pattern as the interface. That's fine if that is intended behaviour, but I just wanted to clarify that the current behaviour for the first-layer raft is not "recitinlear only" -- it adopts the interface pattern.

n8bot, 2020-06-17:
While we're on the topic of supports, I was watching the Prusa live stream from today, and they made an announcement about support features in the alpha version of PrusaSlicer. They allow painting with a brush where supports should be blocked or enforced. This looks like a great feature! Would you have any desire to implement this into SuperSlicer?

supermerill, 2020-06-18:
i'll merge when they are out of alpha/beta

n8bot, 2020-06-20:
I noticed that the option to 'Add settings -> Support material' is available in the Height range Modifier menu, but when selecting it, nothing seems to happen. No added setting to modify.

I was trying to use a Height range Modifier" to change the Support settings for just the first layer, in order to select a different interface pattern, to force a rectilinear first layer raft when using concentric interface for supports. Alas, I can not seem to achieve this. Am I missing a necessary step?

I'm running a build of the master branch from a few days ago.

n8bot, 2020-06-21:
Another aspect of the first layer raft that I wish to bring up, is that when rectilinear grid is chosen as the support pattern, the raft layer is printed parallel, instead of perpendicular to the support material.

Rectilinear:
[image]

Rectilinear grid:
[image]

While I'm posting again, I thought I would post a photo to show what I was talking about in one of my previous comments on this issue: when concentric interface pattern is chosen, the raft layer is concentric as well.

Rectilinear with concentric interface:
[image]

n8bot, 2020-06-21:
I also now notice, that you have chosen to omit the usage of "XY separation between an object and its support" for interface layers. Can we have the option to specify the XY separation for interface layers, which can be different from the XY separation for the main support material?

As well, I've noticed that interface material isn't generated at the "bottom," if supports are building from the model not the build plate. Can we have options to have interface material top and bottom?

Another question arises: Should interface material be placed where incidental XY horizontal contact is made with the support structures, if the XY separation is set to 0? I think that is complicated to implement, but worth thinking about for the future.

Thanks! And also, sorry if I am putting way too many comments into one issue. Please let me know if you'd like me to create individual issues, or make a list of issues and submit them all at one time.

supermerill, 2020-06-21:
> Thanks! And also, sorry if I am putting way too many comments into one issue. Please let me know if you'd like me to create individual issues, or make a list of issues and submit them all at one time.

yeah.. it began to be very complicated.
Also note that the support code will be heavily modified by prusa, so i won't touch before the merge.

n8bot, 2020-06-21:
Sorry for the trouble! Your plan sounds sensible. I'll just write down my current observations privately, and after the merge I will see what remains of my list. Thank you for your work on SuperSlicer!

Nandox7, 2020-07-23:
+1 to having a option to disabled the ""mandatory raft layer under support" (currently mandatory)"
That is currently the most annoying feature in the Slicer for me.

Nandox7, 2020-07-24:
Actually found there is already a PR for PrusaSlicer but it wasn't merged yet.
https://github.com/prusa3d/PrusaSlicer/pull/4280

supermerill, 2020-07-31:
@Nandox7 that pr doesn't remove the raft, it just hides it.

Nandox7, 2020-07-31:
@supermerill ahhh you are totally right. I checked the code and it just adds the option to inflate or not the first layer.
I probably linked the wrong PR or I was too excited about it that I read it wrong. 
This raft is most annoying thing ever in the slicer and it causes more problems than it helps.

supermerill, 2020-07-31:
The thing is, if you don't use the raft/brim (you can choose), the support have a very good chance to peels off the build plate.

Nandox7, 2020-07-31:
I'll check what is done by S3D. I used it for more than 3yrs and never had any issue with loose support.
Only in the case of very small support structures it could be problem and for that I'd use brim.

But the idea is to allow this tas a selectable option, not to remove it completely. So you can still use it if you so wish.

Nandox7, 2020-08-14:
What are those called? I was looking into the code to try and understand what generates them and remove it but not an easy task for someone checking it for the first time.

n8bot, 2020-08-14:
Look at the code from that PR you were talking about https://github.com/prusa3d/PrusaSlicer/pull/4280 and work your way back from there. I don't know myself, I have not closely investigated the code but you should be able to unravel it from there.

Make sure you have VS with Intellisense going -- truly makes it easy to understand codebases.

Nandox7, 2020-08-14:
I read it again and to be honest what it does would already be good, it will only add it under the support pillars.
Tried to merge it but don't see the option in the UI after compiling, need to check why.

n8bot, 2020-08-14:
Not sure, I never compiled a build with that addition myself yet. Personally, to remove the unwanted first layer I simply add a z offset of the first layer thickness and manually delete it from gcode after. This is just for testing before the feature is adjusted.

If you're trying to merge that change int SuperSlicer, there could be other changes that prevent it from working. Off the top of my head, SuperSlicer refers to `tool` instead of `extruder` in some places -- not sure if it is relevant here.

supermerill, 2020-08-16:
@Nandox7 
> don't see the option in the UI after compiling, need to check why.

I'm not using tap.cpp anymore, but the ui files in resources/ui_layout

Nandox7, 2020-08-16:
Thanks all sorted. 
I'm not too familiar with VStudio and wasn't running the proper task to copy over the newly compiled files.
I can see and use that option and it pretty much fixes the annoying raft under support issue.

n8bot, 2020-08-16:
My print with the manually removed raft layer seems to be working fine. [image]

n8bot, 2020-08-18:
Damn, I just had one of the support towers fall! I don't think a tiny little raft would have saved it, though.

What I normally used to do with Simplify3D was make sure there was no individual support islands. In the photo I posted above, you can see slic3r generated 4 islands of support. When I used simplify3d, I would draw the supports in a way to connect all the islands so it was just one big island of support. I think that would have helped save the print in this case.

I think slic3r separates the support into islands in order to be able to adjust the layer heights of each island independently. Is this correct?

n8bot, 2020-08-18:
Actually, maybe the more dense rectilinear support raft from superslicer would have helped save this pillar. However, like I mention in this thread there are issues with supports in superslicer that make them unusable for me. 

:(

[image]

[image]

Nandox7, 2020-08-20:
Can't recall what I used to do with S3D for those cases, but adding brim should helps to give it a wider and stronger adhesion to the bed.

n8bot, 2020-08-20:
S3D does have a more viable brim option for the support structures, but I don't know if a brim alone would have saved this tower. What I used to do in S3D, and I'm trying but failing to replicate in PrusaSlicer/SuperSlicer, is connect all the towers into one contiguous support island. The lateral/horizontal support is often as important as the vertical support, sometimes only for the support structures themselves.
[image]
[image]

With those support structures from S3D, I have successfully printed the part multiple times. The supports never fail.

Nandox7, 2020-08-20:
Know what you mean and just happened to me.

[image]

That support there, that is the only one that is an "island", got loose during the print. Few reasons why that happened but if it was connected to the remaining support it wouldn't have happened.

njoela, 2022-02-01:
I also find the rafts on supports totally useless, time consuming and a waste of filament. This needs to be an option, not mandatory!