supermerill/SuperSlicer #297: Language localization not working and not available in the menu. Only for English

ositobio, 2020-06-15:
There is a problem in SuperSlicer 2.2.51 when you try to change the language. I only shows English.

Using Procmon I got the cause of the issue, and is that the files inside localization\XX are still called Slic3r++ while the new tool looks for SuperSlicer.

WORKAROUND: Rename the files inside your language to SuperSlilcer.po and SuperSlicer.mo (in my case, inside es for Spanish). Start SuperSlicer and choosing language should show up your language. 

[image]
[image]

Hope this helps people and developers to correct it for future versions :)

supermerill, 2020-06-15:
yes, see #273

Anakinsw83, 2020-06-22:
Funciona!

patrickb08, 2020-06-23:
Language Fr does not work by renaming

closed 2020-10-01 by supermerill