supermerill/SuperSlicer #298: 3D Mouse is not detected by the app

altvnk, 2020-06-15:
### Version
`2.2.51` (055cfb80c9f9850456067a03935c8c9ac18cfc42)
### Operating system type + version
macOS Catalina `10.15.5`

### Behavior
* 3D Mouse (3D Connexion Space Navigator) is not detected by **SuperSlicer** while works perfectly fine with **PrusaSlicer**
* While application is in focus, try to move scene using 3D Mouse joystick.
* Scene should react to joystick movement
* Software is not reacting to joystick movement

It may be important that it works on Windows build of SuperSlicer

supermerill, 2020-06-29:
i don't have one to test...

sw3de, 2020-06-30:
I'm using V2.2.52 on a Mac and I can't get my 3D mouse to work, it works in PS and F360.

Is there anything I can do to help?

LaserBee, 2020-07-02:
2.2.52 on Windows 10 here, and my 3DConnextion Spacemouse Pro is working smooth as silk.

supermerill, 2020-07-02:
I'll try to find some time to add some logs to see what's going on.

BeansIsFat, 2020-11-09:
I would also like support for 3D Connexion devices. I confirmed that my device works in PrusaSlicer 2.2.0+ and that it doesn't work in SuperSlicer 2.2.5.3 on Mac OS 10.14.6. If there's anything I can do to help please let me know.

supermerill, 2020-11-10:
@BeansIsFat can you try on the latest nightly? 
https://github.com/supermerill/SuperSlicer/actions/runs/355602784

BeansIsFat, 2020-11-10:
I tried the latest nightly (2.3.55) and it did not respond to the 3Dconnexion device. I noticed that the dialog for controlling options is no longer there either. In the current release (2.2.5.3) Cmd-M opens a 3Dconnexion settings dialog.

supermerill, 2021-05-27:
I made some fixes on the 3D Mouse code.
can someone that have one test it on the latest release (2.3.56) ?

altvnk, 2021-05-27:
It does work now with 2.3.56 on Mac OS 11.4 with 3DxWare 10.7.0
Many thanks for the update!

closed 2021-06-04 by supermerill

denis-itskovich, 2022-10-28:
Not working again in 2.4.58.5