supermerill/SuperSlicer #300: Support for belt printers?

girrrrrrr2, 2020-06-16:
Forgive me if this is the wrong place, but with the newest system of printers being those with a belt as a bed, it would be nice to have the ability to support these printers within the slicer. 

I've seen some post processing scripts and they just seem to take the model/Gcode and just skew them and then go from there. It would be nice to have this motion system supported inside a regularly updated slicer instead of using an old version of Cura or a post processing script.

I plan on making my own belt printer in the near future (from scratch my own design) and i've hit a roadblock pretty early which is finding a slicer that supports this.

supermerill, 2020-06-17:
I can make something but don't have a belted printer.
from what i can understand, it needs 
* a "bed angle" / "extruder angle" and us it to rotate the part before slicing.
* it shouldn't put overhangs where the plate should be.

The most problematic part is the gui. If we want a good rendering, it's a kind of gamble. Mayb i just need to insert a rotation in a line and it's jackpot, but if it's not possible, it's a very big task.

All in all, it may takes me days/weeks to implement it. The user base is extremely small... 
It won't be my priority.

girrrrrrr2, 2020-06-22:
> I can make something but don't have a belted printer.
> from what i can understand, it needs
>
> * a "bed angle" / "extruder angle" and us it to rotate the part before slicing.
> * it shouldn't put [...]

Thats perfectly fine, as long as it's in the pipeline im pretty happy.
The printer hasnt even been designed as much as even started to build.

And you would be one of the first to add this in, so it could be pretty useful for others that arent just me.

One thing id put in when you get around to it is a variable extruder angle, because there are 2 main types currently the 45 degree and the 30-ish degree, so just a variable angle could be pretty handy to encapsulate almost all belt printers.

girrrrrrr2, 2020-08-15:
So I was talking with the guy who made the white knight (a belt printer) and was asking what slicer he uses.

Long story short he said he was willing to help test the code since you dont have a belt printer.

Here is his Twitter https://twitter.com/NAK3DDesigns?s=20
And his youtube is under the same name. 

I know you are probably busy with other stuff. I just thought id let you know.
And the userbase is small... but growing Creality is creating a belt, and there are several DIY belts out there. But everyone just uses blackbelt cura. Which is a modified older version of cura, so... not the most feature rich lol

Moorviper, 2020-08-15:
Only supported Slicer so far ...
Cura Blackbelt
https://github.com/BlackBelt3D/Cura

supermerill, 2020-08-16:
> And the userbase is small... but growing

An affordable and available belt printer is game changer!

girrrrrrr2, 2020-08-16:
Agreed!

NAK3DDesigns, 2020-10-23:
Sorry I wasn't aware of this thread (or forgot if I was told), but I'm willing to help if I can..  One big thing I noticed for the GUI is you need to swap Z & Y when displaying the code (or that appears to be how Blackbelt did it)  this way the infinite axis goes off in the distance not straight down..

supermerill, 2020-10-23:
np
Do you have a release date for the creality printer?
I really want one and I think having one is the kind that will motivate me enough to put his feature on top of the priority list.
I considered making a White-Knight, but i don't have enough free time to build & tinker with it... i already have an other printer kit unopened.

neophyl, 2020-10-23:
Also PS currently has a hard limit on bed size due to max number size I believe.  I remember seeing issues where someone configured a very large bed.  That is why the latest alpha has a max bed limit in it to stop people configuring beds larger than a set value.  Not sure how that will tie in to the infinite axis required in theory.

supermerill, 2020-10-23:
I've revamp all type to push them to double/int64 where needed in susi.
not fully tested, but it should be easier to fix problems if it occurs.

bubnikv, 2020-10-23:
> I've revamp all types to push them to double/int64 where needed in susi.

Thanks for unlocking the true potential of PrusaSlicer to the community. However:

1) boost::polygon Voronoi Diagram does not support int64 coordinates by default https://www.boost.org/doc/libs/1_60_0/libs/polygon/doc/voronoi_advanced_tutorial.htm
2) FillRectilinear2 does neither.

There may be other spots I overlooked.

supermerill, 2020-10-23:
Thanks for pointing it out sensei.
I'll have to create some unit test to see what are the real limits.

I just test it quikly: 
* I created a 2500x2100 bed in PS2.2, and it crashed
* I created the same in myfork, and it can work (even with rectilinear):
[image]
not saying that I "unlocked the true potential", but I just try to push it to the limits i can achieve. You're a company and so you need to have a well tested & robust software. Here it's a bit more the wild west.

girrrrrrr2, 2020-10-23:
> Do you have a release date for the creality printer? (or if I can/it's a good idea to buy a pre-release one)

No information yet on release date has been released.

NAK3DDesigns, 2020-10-23:
> np
> Do you have a release date for the creality printer? (or if I can/it's a good idea to buy a pre-release one)
> I really want one and I think having one is the kind that will motivate me enough to put his feature on top of the priority list.
> I considered making a White-Knight, but i don't have enough free time to build & tinker with it... i already have an other printer kit unopened.

Only info I've been given is a hopeful release date of Mid-November

sioux612, 2020-11-25:
With a few CR30s in the wild now and the german belt printer supposedly going out to backers in the next few weeks as well, the user base and thus possible testers for belt slicers will become existent soon. 

While I would be of very little help in regards to programming anything, I do have both printers on order and would be absolutely willing to beta test/help with anything I can to get support for them in SS.

jhoblitt, 2020-12-08:
I joined the ifactory one belt kickstarter a few weeks ago (regular... not the by Xmas pledge), so I may have one in March if things go well.  Obviously, I'm happy to be an alpha tester.

@supermerill I think you should consider both a fund raiser to help buy you a belt printer and reaching out to see creality and/or ifactory will grant you a machine or at least sell one at a discount.  /CC @sexycyborg

kageurufu, 2021-04-12:
I've got a CR-30 now, and some time to hack around on this. I've got basically no experience with the codebase though

If there's anything I can do to start work towards this, I'd love to help out

supermerill, 2021-04-13:
I tried some little things, like rotating the objects to slice them in a different direction. 
I'm waiting for my code writer rewrite, as the axis are very different, as the y need to be offseted for every z move.
Also it will need a preview adjust to not show the code completely wacky.

You can check it at the belted_printer branch

supermerill, 2021-06-02:
Just get the reality belt printer.
I didn't hope for it before july.
have to work extra hard now.

kageurufu, 2021-06-03:
https://discord.gg/5X3wYxSBHH is the cr-30 discord if you wanna join :)

Gridzilla, 2021-07-01:
+1 would love this feature and willing to help beta test

JFKingsley, 2021-11-11:
Also now got a CR-30 and would love to lend a hand here - where's best to start?

akeigher, 2022-01-04:
TinyBelt from Powerbelt will be launching a kickstarter soon, they are trying to be the "ender3" of belt printers, inexpensive and everywhere. What are the chances that this fork will see development, how can I help (with no coding experience) when I get my belt printer?

supermerill, 2022-01-04:
I have my belt printer in its box, in the middle of my room for too long.
Yet, there is always something more important to do. 
So Hopefully soon(tm)

DocSparky, 2022-01-25:
+1 Just put my CR30 together and melting plastic and would like to be able to use SuperSlicer for it =). I'm willing to be the "better idiot" that crash tests and breaks things.

sisterlimonpot, 2022-02-27:
> I've revamp all types to push them to double/int64 where needed in susi. not fully tested, but it should be easier to fix problems if something happens.
> 
> ```
> Changed precision from ~25b (2 147.48mm -> 0.1um) to ~40b (17km -> 0.1um)
> it's scaled by 10^6 (~20b), with 10^2 (~7b) under the epsilon
> also, CLIPPER_OFFSET_SCALE scale the number by 2^17, and clipper disallow the use of the 2 higher bits, so it can't go higher than 45b
> ```

Where did you put this information? I'm assuming this goes into gcode somewhere in the prusa slicer??? If not, are you willing to walk me through the process? 

If not, at least thanks for considering.

supermerill, 2022-03-02:
> Where did you put this information?

it's just a comment from the source code. You don't have to do anything.

sisterlimonpot, 2022-03-02:
> 

Forgive me when I say that doesn't help. Let me start by saying everything that I read in github seems like a foreign language. I have a cr30 and am trying to find a slicer that is worthy. I'm an ME and software doesn't click for me. 

I'm quite happy with the prusa slicer and found this thread to be a god send. However your comment said that you changed numerical values "from" "to". 

Again, forgive me for assuming, it seems that those changes are made in a source code. Can you provide instruction where to access said source code? Perhaps everything would be made clear if I had that missing piece of the puzzle.

Thanks again 
Jimmy

girrrrrrr2, 2022-03-02:
> Forgive me when I say that doesn't help. Let me start by saying everything that I read in github seems like a foreign language. I have a cr30 and am trying to find a slicer that is worthy. I'm an ME [...]

Currently this is in the alpha stage, where it is being tested so id say its not safe to use yet, and would still recommend using the one given out by Creality until belt support is closer to fully working in SuperSlicer. (just wanting to say thank you to 
SuperMerill for the hard work, we all appreciate you and everything you do)

That being said, if you go here: https://github.com/supermeril/SuperSlicer/tree/belted_printer you can then poke around using the web interface for the files to check out the source code, be warned that its about 15 months old so its behind in several ways along with not being fully complete, otherwise the functionality would be bundled into the main version of SuperSlicer (or a side version possibly).

You can try this: https://github.com/zechyc/Tilted-Bed-Conveyor/
and https://powerbelt3d.com/how-to-orient-models-for-a-conveyor-belt-3d-printer/ and use PrusaSlicer or SuperSlicer to slice and then use that script above to convert the resulting gcode into being usable by the printer.

burner-, 2022-05-13:
Any plans to complete this feature? Creality "own" cura based slicer is just horrifying pile of it... So I really wait that super slic3r start support angled printing.

tablifiedapps, 2022-06-11:
thanks for all you are doing. just like the others, I am eager and willing to help.

jaerixon, 2022-11-14:
Has there been any recent progress on this feature? The SainSmart INFI-20 has bitten me with the belt bug and I'm eager to use SuperSlicer, but no rush. Appreciate all you do for the community!

cartazio, 2022-12-13:
i've some time to do open source hackery this january, what remaining to get this into public alpha/beta shape?

cartazio, 2022-12-13:
i found the relevant source in the ancient blackbelt cura fork https://github.com/BlackBelt3D/Cura/blob/4e18b197ad7cb929058cacdf7ef8031fb6804bbd/plugins/BlackBeltPlugin/StartSliceJob.py#L275

whiskeyblacksmith, 2023-02-03:
The biggest thing for belt printers is that every layer has a "first layer." When printing along the belt, the slicer needs to use the first layer's speed.  The rest of each layer can be at normal speeds.  The rest of the settings are the same as any other printer.

I'm eagerly awaiting this addition to SuperSlicer!  I know time is hard to come by though.

I think the biggest thing holding belt printers up is the lack of slicer adoption.  The Cura Blackbelt slicer is now around 4 years old, KiriMoto is extremely basic and Ideamaker is lacking in advanced features.  Out of the 3 Ideamaker is the best but it is 3-4 years behind SuperSlicer, perhaps more.

Straconis, 2023-06-20:
There haven't really been any updates this year and the link further up for the belted printer fork goes to a 404 now.  Is this still being worked on?

whiskeyblacksmith, 2023-06-20:
> There haven't really been any updates this year and the link further up for the belted printer fork goes to a 404 now. Is this still being worked on?

https://github.com/supermerill/SuperSlicer/releases

In early March the developer said that they were going to be working on support for treadmill (belt) printers.  They said a version should be released in April for testing.  I'm guessing that life got in the way and it got delayed.  There is hope!

JohnySaak, 2023-11-03:
I hope for the best. Using other slicers like creality slicer or idea maker is driving me nuts :D
Do we have any updates ? 
Also i willing to pay for this. Some kind of go fund could help.  User base is but hardcore.

supermerill, 2023-11-23:
I have a branch (https://github.com/supermerill/SuperSlicer/tree/belted_printer) with half-working thing.
need to work on the gui a bit, and make more tests.

whiskeyblacksmith, 2023-11-23:
@supermerill this is fantastic news!  Thank you so much for all your hard work!  I will test it as soon as I can and report back.

Colinmanlives, 2023-11-23:
Thank you so much ideamaker is infuriating with support materials

JonasZaverka, 2023-11-23:
Man, this made my day 👍 
Is there a Prebuild available for testing or do i need to build it myself ? 

Just cant wait to slice something <3

martinjohuber, 2023-11-25:
I tried to compile this but can't get it working, I followed the guide but always get an error in visual studio (Code -1 (0xffffffff)). 

I am not an experienced programmer, did anyone have success to compile this and can provide a prebuild, please? 

I also can't wait to test this!

whiskeyblacksmith, 2024-02-27:
@supermerill I have been unable to compile the belt version due to errors.  @martinjohuber seems to have the same or similar issues.  I tried reinstalling Visual Studio on two different computers with no luck and unfortunately, this type of programming is not my strong suit.  Do have any advice on getting this to compile or is there a chance we can get a precompiled version to test?

Lyude, 2025-01-08:
Hey! Figured I'd ask since it seems like the last comment was on Feb 26 of last year, I assume this is still a WIP?
I recently bought myself a baby belt but did not realize there aren't really any decent open source not-in-a-browser slicers available, so if there's still interest in getting this into the mainline branch at some point I'd love to give it a try. happy to build myself as well

the3dfox, 2025-01-09:
we have multiple threads now for this topic: https://github.com/SoftFever/OrcaSlicer/issues/6885#issuecomment-2571803292

martinjohuber, 2025-03-02:
Somebody made a campaign to get this going: [](https://www.gofundme.com/f/get-3d-belt-printers-into-orca-slicer ) the code from Orca SLicer would for sure also help SuperSlicer to implement this as well