supermerill/SuperSlicer #301: Issues compiling on Ubuntu

PlantLily, 2020-06-17:
### Version
latest clone from github

### Operating system type + version
Ubuntu 18.04.1, Linux kernel 5.3.0-59-generic

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
Trying to compile on Ubuntu, built deps and then ran `cmake .. -DSLIC3R_STATIC=1 -DCMAKE_PREFIX_PATH=/home/lily/.SuperSlicer/deps/build/destdir/usr/local -SLIC3R_BUILD_TESTS=OFF` after navigating to the build directory, and then adding add_compile_options(-fpermissive) to the main CMakeLists and changing the înfill_peri_overlap to infill_peri_overlap in src/libslic3r/PerimiterGenerator.cpp to resolve errors from the weird î character, i now get the following message:
 `make[2]: *** No rule to make target '../deps/build/destdir/usr/local/lib/libwx_gtk2u_scintilla-3.1.a', needed by 'src/superslicer'.  Stop.
CMakeFiles/Makefile2:216: recipe for target 'src/CMakeFiles/slic3r.dir/all' failed
make[1]: *** [src/CMakeFiles/slic3r.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
`
TIA!

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

supermerill, 2020-06-17:
yeah, i put î instead of a i by error.
It's corrected in my dev branch.
For compiling, i suggest you to reset hard to the tag 2.2.51.1 (commit c68e94daf2339888ae8f9e5b196075ea00bd5af0 )

supermerill, 2020-06-17:
I've updated the master branch. But i didn't test it on ubuntu yet, so there is a good chace that it doesn't compile for you.
I've also added a "latest_release" branch, so you can checkout this easily.

closed 2020-07-03 by supermerill