supermerill/SuperSlicer #302: Extra perimeters on Odd layers still generated inside a modifier set to 0 perimeters

neophyl, 2020-06-17:
### Version
2.2.51

### Operating system type + version
Win10 64bit

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Been doing some experimenting with the bridge width amongst other things trying to get nicer bottom surfaces on those when I found a different issue.  One new thing that I love is the extra perimeters on odd layers to tie in the infill more.  Its a great feature.  However I added a modifier to a test part to stop it printing an area (zero perimeters, top/bottom and infill).  When you do that though you still get the extra perimeters on odd layers generated.
I've attached a project with the issue configured.

_Is this a new feature request?_
No

[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/4794806/Shape-Box.zip)

supermerill, 2020-06-18:
If i disable the option, people that may want it even with 0 perimeters may be angry. Don't you think?

neophyl, 2020-06-18:
I dont know how other people would use it.  As they dont attach to anything though then printing them is pointless, at least in this case.  
About the only use case for them I could see is for artistic purposes where you have infill but 0 normal perimeters.  That might give an interesting effect.
How hard would it be to add that setting into those available in modifier ?

supermerill, 2020-06-18:
> How hard would it be to add that setting into those available in modifier ?

extra perimeter on odd layer should already be available in modifiers, unless you think about something else?

neophyl, 2020-06-18:
Sorry didn't realise the tickbox was available in there.  Had to turn the perimeters to 1 to un-tick it before putting it back to zero but that got rid of the floating perimeters.  
As configuring things properly removes the issue I'll close this and I apologise  for wasting your time :)

supermerill, 2020-06-18:
I'll do something to make it enabled even if perimeters are at 0