supermerill/SuperSlicer #304: Inconsitent slicing on edges

uepsie, 2020-06-20:
### Version
2.2.51

### Operating system type + version
Win10 64Bit - 1909 (18363.900)

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 + Klipper

### Behavior
Slicing a benchy from the original stl provided by 3dbenchy.com - default settings for Prusa MK3, Prusa PLA and Quality 0.15mm

It seems that the layer vary at edges (one vertex)
it is visible in the slicing preview and also on the model:
[image]

It seems to start at the same height as the vessel's side starts to rise and ends at the top of the box.
but there are other inconsistencies at the same edge.

From comparing the layers with a gcode viewer you can see that the curve geometry seems to change between layer 10.85mm and 11mm
[image]

even though there's no change in the stl triangles
[image]

n8bot, 2020-06-20:
This is due to a built-in limitation to the libslic3r library.

In the file libslic3r.h, you can find [this line](https://github.com/supermerill/SuperSlicer/blob/a60631b7aa1ce28b5c90518a00b8996ba6024130/src/libslic3r/libslic3r.h#L38): ```#define RESOLUTION 0.0125```

(That is as it stands in current master branch of SuperSlicer. Current version of slic3r and aplha versions of PrsuaSlicer have a different line, using ```constexpr```.)

The variable represents the minimum value of the Slicing setting "resolution," even if that setting is set to 0 in GUI. I have built custom versions of Slic3r, PrsuaSlicer, and SuperSlicer, replacing ```0.0125``` with ```0.001``` with great success. No more misaligned layers, no more simplification of meshes (beyond 1 micron simplification). Try it! If you'd like, I can share my windows binaries with the change implemented already (as well as increasing E_NUM precision to 6, which allows 6 digits of E output instead of 5).

The one major downside of making this modification, is that the "secondary" 3D gcode visualization begins having errors, with parts of the toolpaths missing completely from screen. The underlying g-code is unaffected, though. It's just the visualization.

n8bot, 2020-06-20:
I forgot to mention, in combination with the change I made to libslic3r.h, I had to set "Model rounding precision" to zero in order to achieve smooth layers. If that value retains its default value of 0.0001, there were tiny layer misalignments remaining, even with ```#define RESOLUTION 0.001``` in libslic3r.h.

supermerill, 2020-06-21:
I tried multiple times to put the resolution as a setting but it creates bugs. not as easy as it sounds.

uepsie, 2020-06-21:
@n8bot thanks a lot
I am going to try this tonight

@supermerill Thanks for your amazing work! You should accept donations :-)

n8bot, 2020-06-21:
> I tried multiple times to put the resolution as a setting but it creates bugs. not as easy as it sounds.

What bugs did you see? I am only experiencing missing segments in the gcode visualization, but only after the gcode file itself is reloaded. Inspecting the gcode reveals no missing segments. Were there other errors I should be aware of?

supermerill, 2020-06-21:
> @uepsie Thanks for your amazing work! You should accept donations :-)

there is a "sponsor" button on the top of the page if you want to.

Erider-yoann, 2020-09-08:
I am extremely interested in resolving this issue because I am regularly confronted with it on many projects and the imprecision ruins some parts.

[image]

Unfortunately I didn't manage to compile SuperSlicer by modifying libslic3r.h as suggested by n8bot due to lack of knowledge about compiling, installing libraries (especially Boost) etc ...

In any case thank you @supermerill for all the time and energy you put in this project ! Love SuperSlicer !

qwewer0, 2021-06-02:
@uepsie Could you please test it on the latest release?

uepsie, 2021-06-03:
works perfectly fine now with 2.3.56.1