supermerill/SuperSlicer #306: No gap fill independent of settings

SnowDrifterr, 2020-06-21:
### Version
_2.2.51_

### Operating system type + version
_Win 10 1909_

### 3D printer brand / version + firmware version (if known)
_Duet RRF 2.5.1_

### Behavior
* _Gap fill not rendered in settings, reflected in physical print_
* _slice a model with slowly varying thickness, gap fill areas are shown as blank_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
have gap fill
* _Actual Results_
have no gap fill
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs

[covid19_headband_smooth.zip](https://github.com/supermerill/SuperSlicer/files/4808813/covid19_headband_smooth.zip)

Wil-Storch, 2020-06-21:
Change your infill patterns from Rectilinear to Rectilinear (Filled). These regions do not fill otherwise as a standard perimeter width doesn't fit

supermerill, 2020-06-21:
gap fill speed is set to 0. 
it was the old way to disable gap fill, i kept it to avoid breaking too much profile.

SnowDrifterr, 2020-06-21:
Ahh

Well now this brings up another behavior. Setting gap fill to an arbitrary value (40mm/s) with 3 perimeters seems to break auto speed and sets the whole print to go extremely slow in the 10-12mm/s range. 

Reducing number of walls to 2, or changing speeds manually seems to resolve it

supermerill, 2020-06-21:
from the code:

	        // In order to honor max_print_speed we need to find a target volumetric
	        // speed that we can use throughout the print. So we define this target 
	        // volumetric speed as the volumetric speed produced by printing the 
	        // smallest cross-section at the maximum speed: any larger cross-section
	        // will need slower feedrates.

so in the current algorithm, i think it search the thinnest extrusion ( and gapfill can create very thin ones) and it slow down everything else to have the same volumetric flow than with very thin gap fill at 50mm/s.

I'll have to rewrite completly the algorithm i think. For the moment, i'll just remove gapfill/ thinwalls from it.

VanessaE, 2020-06-24:
> For the moment, i'll just remove gapfill/ thinwalls from it

Autospeed is currently broken entirely, possibly because of that change. :confused:  

For example: load this config, [superslicer-config-2020-06-24.ini.zip](https://github.com/supermerill/SuperSlicer/files/4823458/superslicer-config-2020-06-24.ini.zip) and this model, https://www.thingiverse.com/thing:3630709 (the harvester, not the egg).  Slice it, then look in the preview:  you'll find that everything's been forced to ~1.1 mm/s, *except for gap fill*, which is at the configured speed (200 mm/s).

In my opinion you should do what upstream Slic3r does:  a checkbox to en/disable gap fill, with either an exact speed or "auto".  Then just print gap fill lines at whatever speed it takes to meet the volumetric target, like you would any other feature.

BUT, go a step further:  if a gap fill line is so thin that its theoretical speed exceeds the volumetric `max_print_speed`, cap it at the max, and then *ignore* its speed when calculating the speeds of other features.  That way the other features still print at close to their volumetric target.

VanessaE, 2020-06-24:
> For the moment, i'll just remove gapfill/ thinwalls from it

Autospeed is currently broken entirely (commit 5d1aff62a46cfb124a8a9d6fbe95286466f6445b), possibly because of that change. :confused:  

For example: load this config, [superslicer-config-2020-06-24.ini.zip](https://github.com/supermerill/SuperSlicer/files/4823458/superslicer-config-2020-06-24.ini.zip) and this model, https://www.thingiverse.com/thing:3630709 (the harvester, not the egg).  Slice it, then look in the preview:  you'll find that everything's been forced to ~1.1 mm/s, *except for gap fill*, which is at the configured speed (200 mm/s).

In my honest opinion, you should follow upstream Slic3r's lead here:

As you know, it currently uses a checkbox to enable gap fill, and allows "Auto" for all speeds (at least other than layer 1), including for gap fill.  With gap fill enabled and set to "Auto", each such line segment just prints however fast it takes to meet the target flow specified by `max_volumetric_speed`, and is capped at `max_print_speed`.

Having only 0-or-speed for the gap fill field precludes the notion of anything other than a fixed speed; setting it to 0 to disable it is actually very annoying, as this is precisely the purpose of a checkbox.  As for breaking profiles, surely you could convert it, i.e. if it's set to 0 in the file, then change it to "Auto" (or to the value the `- default -` profile would have used), turn off the new checkbox, and offer to re-save the profile.

Either way, it's trivial to use a bit of `sed` magic to fix all profiles at once (I do this from time to time, when I want to change something in a bunch of my profiles to the same value all in one go).

BUT you should go a step further past Slic3r:

Slicer has a glitch where if a gap fill line segment speed was capped, it will print at somewhere below the user's volumetric target speed, by definition, and that kills the speeds of everything else on the layer.  I.e. the thinnest gap fill line segment on a layer effectively sets the flow for everything on that layer, so the thinner that line segment is, the slower the whole layer will print.

SO, if a gap fill line segment has to be capped at the configured max speed, *ignore* that segment's speed when calculating the speeds of other features (including other gap fill lines that *are* thick enough to not hit the cap).  That way, everything else will print at close to the volumetric target even if some gap fill lines can't, and the whole print will display more consistent speeds from layer to layer.

Prusaslicer kept the original "0's everywhere" method, so gap fill screws up autospeed's calculations there, as well.  Besides, using to "0" to mean "automatic" when the field is meant to be a specific linear speed triggers my OCD. :wink:

supermerill, 2020-06-24:
> Autospeed is currently broken entirely

Seems like i forgot that the gapfill may form a loop, corrected in next commit.

> Having only 0-or-speed for the gap fill field precludes the notion of anything other than a fixed speed; setting it to 0 to disable it is actually very annoying, as this is precisely the purpose of a checkbox

Yeah, it's maybe about time to remove that double kill-switch, and let the current checkbox 'alone'. Note that it means that with 0, it will get the 

> Slicer has a glitch where if a gap fill line segment speed was capped, it will print at somewhere below the user's volumetric target speed, by definition, and that kills the speeds of everything [...]

Duplicate of #306, already fixed

VanessaE, 2020-06-24:
> Duplicate of #306, already fixed

Oh for crying out loud, how did I miss that? :stuck_out_tongue_closed_eyes:

closed 2020-08-24 by supermerill