supermerill/SuperSlicer #307: [Question] Support Z-offset in % of extruding witdh

robotimmy, 2020-06-21:
### Version
2.2.51

Why is Z-offset in % (in Support material options) defined as "% of extruding width for the interfase layers" and not "% of the base layer height"?
I think it makes more sense to use the base layer height (or current layer height if using variable layer height). 

For example if using base layer height as reference:
- Z-offset = 100%
- base layer height = 0.2mm => gap between interface layer and object layer is 0.2mm
- base layer height = 0.05mm => gap between interface layer and object layer is 0.05mm

in v2.2.51 gap is always the extrusion width (here 0.4mm) with different layer heights.

[image]

supermerill, 2020-06-21:
Imo, the gap shouldn't vary if you decide to change the layer height, as a bridge is always the same diameter.
I can't use the nozzle diameter directly, because you can have multiple ones. That's why i decided for using the width.

robotimmy, 2020-06-21:
Thank you for quick response.
I usually set the Z-offset to base layer height like in Simplify3D. There you can select the number of seperation layers in support options. So there is always one empty base layer between support interface and object to get imo best results ("easy" to remove supports + best surface).
If you slice with 0.3mm layer height z-offset is 0.3mm and if you slice with 0.1mm layer height z-offset is 0.1mm.
The problem was that I sliced an object with 0.05mm layer height, but forgot to change the z-offset (it was was 0.32mm) and the print failed. Therefore the question.

supermerill, 2020-06-21:
why does it fail?

robotimmy, 2020-06-21:
There was no layer bonding between interface and object. And the sharp edge was printed in mid-air.
But I think I found the problem. I forgot to select "Thin walls". So the gap between interface layer and object was about 1.6mm...

[image]

robotimmy, 2020-06-22:
solved

closed 2020-06-22 by robotimmy