supermerill/SuperSlicer #308: cmake can't complete: wxWidgets not found

VanessaE, 2020-06-22:
### Version

Just cloned, git commit 66847ebfa022c37a02666e7bafb6016b793cc9ca

### Operating system type + version

Debian buster

### 3D printer brand / version + firmware version (if known)

Custom Hypercube running Marlin

### Behavior

Tried to clone and compile today just to play around with it, but the `cmake` step fails when it tries to reference wxWidgets, claiming it is not installed.  3.1 **is** installed, and I can compile Prusaslicer against it without a problem.

<details><summary>Full cmake output</summary>

```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: 1
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: OFF
-- CMAKE_PREFIX_PATH: (default)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.64.0") found components: system filesystem thread log locale regex chrono atomic date_time 
-- Boost::boost exists
-- Found TBB: /usr/include (found version "2020.2")  
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = /usr/include
--   TBB_DEFINITIONS         = 
--   TBB_LIBRARIES           = /usr/lib/x86_64-linux-gnu/libtbb.so
--   TBB_DEFINITIONS_DEBUG   = TBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = 
--   TBB_DEFINITIONS_RELEASE = 
--   TBB_LIBRARIES_RELEASE   = /usr/lib/x86_64-linux-gnu/libtbb.so
-- Found CURL: /usr/lib/x86_64-linux-gnu/libcurl.so (found version "7.68.0") 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
[... 641 lines cut ...]
  return 0;
}

```
</details>

As with Slic3r, Prusaslicer, and anything else I compile from sources, I created a script to simplify it:

<details><summary>My build script</summary>

```bash
#!/bin/bash

cd /home/vanessa/RepRap/SuperSlicer/

rm -rf build
mkdir build
cd build
cmake .. -DSLIC3R_FHS=1 #-DSLIC3R_WX_STABLE=1
make -j8
sudo make install
```
</details>

If I enable `-DSLIC3R_WX_STABLE=1`, it naturally fails to build, since I don't have wx 3.0 installed.

supermerill, 2020-06-22:
i don't know. I build the deps every time when i'm building on linux.
here is my script:
```
sudo apt update
sudo apt install libgtk2.0-dev libglew-dev libudev-dev
mkdir deps/build
cd ./deps/build
cmake ..
make
cd ../..
mkdir build
cd build
cmake .. -DCMAKE_PREFIX_PATH="/home/runner/work/SuperSlicer/SuperSlicer/deps/build/destdir/usr/local" -DSLIC3R_STATIC=1
make slic3r
mkdir package
mkdir package/bin
cp -Rf ../resources package/resources
cp -f src/superslicer package/bin/superslicer
echo -e '#!/bin/bash\nDIR=$(readlink -f "$0" | xargs dirname)\nexport LD_LIBRARY_PATH="$DIR/bin"\nexec "$DIR/bin/superslicer" "$@"' >superslicer
chmod u+x superslicer
```
you'll have to change the DCMAKE_PREFIX_PATH, and you should be able to use the same as with ps.

VanessaE, 2020-06-22:
That differs greatly from the build instructions in the documentation (https://github.com/supermerill/SuperSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md).

I already have the mentioned -dev packages installed.  I don't build Prusaslicer statically though; it builds fine in the normal/dynamic way like other stuff. Here's my script for that:

```bash
#!/bin/bash

cd /home/vanessa/RepRap/PrusaSlicer/
cd build
cmake .. -DSLIC3R_FHS=1
make -j8
sudo make install
```

I use your method, building SuperSlicer statically, it gets a lot further into the build... The deps build without obvious errors, and `cmake` completes fine as far as I can tell.  The final `make slic3r` step compiles for a bit, but dies about a third of the way through:

```make
[ 29%] Built target libslic3r_cgal
[ 29%] Precompiling pchheader.hpp for libslic3r (C++)

[... a bunch of warnings that are probably inconsequentual ...]

In file included from /home/vanessa/RepRap/SuperSlicer/build/src/libslic3r/libslic3r_pch/pchheader.hpp:113:
/home/vanessa/RepRap/SuperSlicer/src/libslic3r/Config.hpp: In member function ‘double Slic3r::ConfigOptionPercentsTempl<NULLABLE>::get_abs_value(size_t, double) const’:
/home/vanessa/RepRap/SuperSlicer/src/libslic3r/Config.hpp:870:87: error: there are no arguments to ‘is_nil’ that depend on a template parameter, so a declaration of ‘is_nil’ must be available [-fpermissive]
  870 |     double                  get_abs_value(size_t i, double ratio_over) const { return is_nil(i) ? 0 : ratio_over * get_at(i) / 100; }
      |                                                                                       ^~~~~~
/home/vanessa/RepRap/SuperSlicer/src/libslic3r/Config.hpp:870:87: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/home/vanessa/RepRap/SuperSlicer/src/libslic3r/Config.hpp:870:116: error: there are no arguments to ‘get_at’ that depend on a template parameter, so a declaration of ‘get_at’ must be available [-fpermissive]
  870 |     double                  get_abs_value(size_t i, double ratio_over) const { return is_nil(i) ? 0 : ratio_over * get_at(i) / 100; }
      |                                                                                                                    ^~~~~~
make[3]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:67: src/libslic3r/libslic3r_pch/pchheader.hpp.gch/.c++] Error 1
make[2]: *** [CMakeFiles/Makefile2:1328: src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:725: src/CMakeFiles/slic3r.dir/rule] Error 2
make: *** [Makefile:201: slic3r] Error 2
```

<details><summary>Full build log</summary>
```
vanessa@rainbird:~/RepRap/SuperSlicer/build$ make slic3r -j8
Scanning dependencies of target hidapi
Scanning dependencies of target encoding-check
Scanning dependencies of target clipper
Scanning dependencies of target admesh
Scanning dependencies of target nowide
Scanning dependencies of target miniz_static
Scanning dependencies of target avrdude-conf-gen
[  0%] Building C object src/hidapi/CMakeFiles/hidapi.dir/linux/hid.c.o
[  0%] Building CXX object src/build-utils/CMakeFiles/encoding-check.dir/encoding-check.cpp.o
Scanning dependencies of target glu-libtess
[  1%] Building C object src/miniz/CMakeFiles/miniz_static.dir/miniz.c.o
[  2%] Building CXX object src/avrdude/CMakeFiles/avrdude-conf-gen.dir/conf-generate.cpp.o
[  2%] Building CXX object src/clipper/CMakeFiles/clipper.dir/clipper.cpp.o
[  2%] Building CXX object src/boost/CMakeFiles/nowide.dir/nowide/iostream.cpp.o
[  2%] Building C object src/glu-libtess/CMakeFiles/glu-libtess.dir/src/dict.c.o
[  2%] Building C object src/glu-libtess/CMakeFiles/glu-libtess.dir/src/geom.c.o
[  2%] Building C object src/glu-libtess/CMakeFiles/glu-libtess.dir/src/memalloc.c.o
[  3%] Building C object src/glu-libtess/CMakeFiles/glu-libtess.dir/src/mesh.c.o
[  3%] Linking C static library libhidapi.a
[  3%] Building C object src/glu-libtess/CMakeFiles/glu-libtess.dir/src/normal.c.o
[  3%] Building CXX object src/admesh/CMakeFiles/admesh.dir/connect.cpp.o
[  3%] Building C object src/glu-libtess/CMakeFiles/glu-libtess.dir/src/priorityq.c.o
[  3%] Built target hidapi
Scanning dependencies of target polypartition
[  4%] Building CXX object src/polypartition/CMakeFiles/polypartition.dir/polypartition.cpp.o
[  4%] Linking CXX executable avrdude-conf-gen
[  6%] Linking CXX static library libnowide.a
[  7%] Building C object src/glu-libtess/CMakeFiles/glu-libtess.dir/src/render.c.o
[  7%] Built target nowide
[  8%] Linking CXX executable encoding-check
[... 333 lines cut ...]
  113 |             bool stack_unwinding_in_progress = std::uncaught_exception();
      |                                                                        ^
In file included from /usr/include/c++/9/new:40,
                 from /usr/include/c++/9/bits/stl_construct.h:59,
                 from /usr/include/c++/9/bits/stl_tempbuf.h:60,
                 from /usr/include/c++/9/bits/stl_algo.h:62,
                 from /usr/include/c++/9/algorithm:62,
                 from /home/vanessa/RepRap/SuperSlicer/build/src/libslic3r/libslic3r_pch/pchheader.hpp:25:
/usr/include/c++/9/exception:102:8: note: declared here
  102 |   bool uncaught_exception() _GLIBCXX_USE_NOEXCEPT __attribute__ ((__pure__));
      |        ^~~~~~~~~~~~~~~~~~
[... cut ...]
  870 |     double                  get_abs_value(size_t i, double ratio_over) const { return is_nil(i) ? 0 : ratio_over * get_at(i) / 100; }
      |                                                                                                                    ^~~~~~
make[3]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:67: src/libslic3r/libslic3r_pch/pchheader.hpp.gch/.c++] Error 1
make[2]: *** [CMakeFiles/Makefile2:1328: src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:725: src/CMakeFiles/slic3r.dir/rule] Error 2
make: *** [Makefile:201: slic3r] Error 2
```

supermerill, 2020-06-22:
yeah, i didn't build on linux since tag 2.2.51.1, there may have some things that vc++ likes but not g++.

supermerill, 2020-06-24:
should be good now. I had some problem with the wxwidgetscintilla lib, had to renamed it.

VanessaE, 2020-06-24:
Yep looks fine now.  I can indeed compile the "normal" way, without pre-building the "deps" or any static stuff, as below (after I nuked the folder and re-cloned), in the same  manner as with PS, and run it from `./build/src/superslicer`.

One variation:  I still do `make slic3r -j8` here, as you suggested earlier.  You might want to update the build doc.

Btw, @supermerill you never answered my question on that other Prusaslicer issue regarding thin walls :smile:

closed 2020-06-24 by VanessaE