supermerill/SuperSlicer #309: gcode comment for outer perimeters

JoduWei, 2020-06-22:
### Version
2.2.51

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
All

### Behavior
I'd like to use some post processing action and I need place holders for outer perimeters. :)
I think it would be a very small change.

supermerill, 2020-06-23:
what you you mean by outer ? it's the external perimeter? and the external ones on the holes?
[image]
you talk about a different comment string when using the verbose output, right?

n8bot, 2020-06-23:
> what you you mean by outer ? it's the external perimeter? and the external ones on the holes?

Yes, I think that is what is meant.

> you talk about a different comment string when using the verbose output, right?

I think it's important to have the External perimeter distinguishable in the custom gcode, as is described in this comment on another issue: https://github.com/supermerill/SuperSlicer/issues/237#issuecomment-627982783

I don't know if verbose gcode uses the same mechanism.

JoduWei, 2020-06-23:
Yea i mean the comment strings in verbose gcode.
Something like this:
`G1 X123.143 Y98.643 ; move to external outer perimeter point
G1 F2715
G1 X126.857 Y98.643 E0.16408 ; outer external perimeter
G1 X126.857 Y99.143 E0.02209 ; outer external perimeter

G1 X123.143 Y98.643 ; move to external inner perimeter point
G1 F2715
G1 X126.857 Y98.643 E0.16408 ;  inner external perimeter
`

supermerill, 2020-06-23:
I'll do something with the information available at the gcode generator.

JoduWei, 2020-06-23:
ty :)