supermerill/SuperSlicer #311: Wall thickness in addition to perimeters

guestisp, 2020-06-23:
Would be possible to add a wall thickness parameter in addition to perimeter number ?

In exaple, when printing with a 0.40 nozzle and extrusion width (0.40), 3 perimeters means 1.20mm.
If you change the nozzle size to a 0.80 or bigger, with extrusion width set to 1.2, 3 perimeters means 3.60mm but in most cases, would be OK to print the same object with just 1 periemter (1.20mm) in much less time

So, if wall thickness is set, it should override the perimeter number, if left to 0, perimeter number should be considered.

This allow the user to change nozzle and extrusion width but keeping the same printed thickness.

yschroeder, 2020-06-25:
This would be a very nice addition! Some more info also about other issues this feature would address can be found here:
https://github.com/prusa3d/PrusaSlicer/issues/4019

More specifically I already posted my idea of this feature there:
https://github.com/prusa3d/PrusaSlicer/issues/4019#issuecomment-623184196

It boils down to this: Actually, nobody really cares about the number of perimeters of an object. Instead, one should be able to configure the desired wall thickness. Needed parameters are:

On Printer Settings tab:
* `Minimum extrusion width` (already present in SuperSlicer, basically your nozzle diameter)
* `Maximum extrusion width` (depends on nozzle geometry, can be large e.g. for E3D nozzles with flat surface)

On Print Settings tab:
* `Target wall thickness`

The slicer then tries to ensure the `target wall thickness` everywhere with the **minimum number of perimeters**. If the space is too narrow between two outer perimeters (thin walls) it reduces the extrusion width (this feature is already present). If the space between two walls is a bit too wide and would require gap fill, it increases the extrusion width slightly to avoid the gap fill operation.

supermerill, 2020-10-01:
Proposal:
* add a Maximum extrusion width in extruder
  * the real extrusion width is min(extrusion width as defined in width tab, Maximum extrusion width)
* add a Target wall thickness (in mm) near number of perimeters
  * will deactivate external perimeter width, perimeter width
  * will use at least "perimeters" perimeter (it will fire a popup if there are too many, to be sure if it's the user intent)
  * will set custom perimeter extrusion widths & perimeter number for each object (or region if using a different extruder).

objective: if you're printing a box with a 1.5 mm wall, you'll get the best result in one click.

yschroeder, 2020-10-02:
You said it would be done per object. It would work for a box with 1.5mm walls on all sides, but would fail for a box with one wall being a different thickness, right? This would also not work correctly for the cog wheels mentioned in https://github.com/prusa3d/PrusaSlicer/issues/4019

I think this needs to be evaluated per layer/line segment (and that sounds very computationally expensive).

Let me try to understand your proposal with an example:

### Settings

* Nozzle size: 0.4mm
* Maximum extrusion width: 1mm (flat E3D nozzle)
* Extrusion width: 0.45mm (reasonable default for 0.4mm nozzle)
* Target wall thickness: 2.5mm (I want a strong print)
* Perimeters (minimum): 2

### Test cases
Note: I omit the overlaps for the extrusions to bond together in my calculations.

#### Case 1
* Object: Solid object
* Result: 3 perimeters each 0.83mm (`2.5mm / ceil(2.5mm / 1mm)`)
or
* Result 5 perimeters each 0.5mm (`2.5mm / floor(2.5mm / 0.45mm)`)

#### Case 2
* Object: Hollow object (vase) 0.3mm wall thickness
* Result: 1 perimeter (this would be below the minimum) of 0.45mm width (results in overextrusion)

#### Case 3
* Object: Hollow object (vase) 0.5mm wall thickness
* Result: 1 perimeter (this would be below the minimum) of 0.5mm width

#### Case 4
* Object: Hollow object (vase) 1.5mm wall thickness
* Result: 2 perimeters each 0.75mm (`1.5mm / ceil(1.5mm / 1mm)`)
or
* Result: 3 perimeters each 0.5mm (`1.5mm / floor(1.5mm / 0.45mm)`)

I as you can see, there are two possible solutions for two of the tests cases. I would always prefer the solution with less but thicker perimeters as they are stronger and generally print faster with better surface finish.

Also there is two edge cases: bridges and overhangs.

For bridges the perimeter width (and spacing) should be the nozzle size to avoid gaps, see #407 

For overhangs, the width of the external perimeter should be chosen, so that at least 50% of it is supported by the layer below (to avoid printing the external perimeter in thin air). This means for steeper overhangs, the external perimeter should be thinner. For slight overhangs it can be thicker.

supermerill, 2020-10-02:
I said "per object / region" only because you can assign a different extruder (and so nozzle) / perimeter width to it.

before slicing, it will initialize the perimeter count & width with a formula something like:
```
count = min(perimeters, 1+ (peri_obj_width -  ext_peri_width) / to_spacing(peri_width))
peri_width = max(peri_width, min(max_width, to_width( (peri_obj_width -  ext_peri_width) / count ) ))
ext_peri_width = max(ext_peri_width, min(max_width, peri_obj_width - count *peri_width))
```

and nothing else.

yschroeder, 2020-10-02:
I think I misunderstood what all this is about. Also I am not able to follow your formulas, as I don't know what `to_spacing` does and some of the variables are.

My experience is that it is not sufficient to adjust only perimeters to the extrusion width of a different nozzle, but you also need to change infill percentage, as the infill width changes as well and all the infill patterns rescale dramatically with changing extrusion width.

supermerill, 2022-07-26:
I added a "wall thickness". It doesn't compute the perimeter number but the perimeters width.
Is it good enough or not?

yschroeder, 2022-07-26:
I guess that would work. However, with Arachne perimeters in prusa slicer 2.5 a lot of perimeter related issues become obsolete...

supermerill, 2022-07-27:
> with Arachne perimeters a lot of perimeter related issues become obsolete...

Depends on your model, I made some tests and can be very suboptimal.

CCS86, 2022-07-27:
Yeah, the "stepped" nature of the arachne engine is disappointing. Is there a functional reason the steps couldn't be much smaller and more dense?

guestisp, 2022-10-19:
Honestly, i'm using the arache version of SuperSlicer and i didn't find anything strange except some little underextrusion (i think it's a bug) solved by using 1.2 as extrusion multiplier. Having no gap-fill is priceless and makes printing MUCH easier avoiding that micro-extrusion done by gapfill

I can't wait for the "fixed" arachne version, as most of perimeters related things are not working anymore (like 1 periemeter on top surface). Other than that, i also don't care the perimeters number, I would like to set the wall thickness and stop (like the bottom and top thickness) but it doesn't work.