supermerill/SuperSlicer #312: "Only one perimeter on top surfaces" causes artifacts on top slopes

Czarnyszarf, 2020-06-23:
### Version
Version: 2.2.51

### Operating system type + version
OS: Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Printer: Prusa MINI , stock build-in profile with ironing disabled

### Behavior
One perimeter on top causes unnecessary infill on sloping surfaces, it makes "fuzzy" skin on them during printing.  As I see - on each layer it makes "solid infill" winch causes over extrusion.

I attached 3MF file wnich shows you the issue , you can examine it closely :)

Expected results: 1 perimeter only on the very top flat surfaces .

[3MF file showcase of the problem.zip](https://github.com/supermerill/SuperSlicer/files/4818455/3MF.file.showcase.of.the.problem.zip)

One perimeter on top(enabled/disabled) : 
[image]
[image]

neophyl, 2020-06-23:
Duplicate of #248 I think.

Czarnyszarf, 2020-06-24:
thanks for the solution!

Tested it and it seems 200-250% is best value for normal 0.4 nozzle

supermerill, 2020-06-24:
ok, i'll change the default value to 200.

closed 2020-08-24 by supermerill