supermerill/SuperSlicer #313: Time Estimation Overflow

thierryzoller, 2020-06-24:
- Download attached
- Slice
- See Time estimation

[overflow.zip](https://github.com/supermerill/SuperSlicer/files/4825040/overflow.zip)
[image]

supermerill, 2020-06-24:
[image]

thierryzoller, 2020-06-24:
This happens sporadically and I don't know why. I had minimised the window, sliced again and it estimated correcty now...

thierryzoller, 2020-06-24:
2147483647 is the max value of a 32 bit integer, so it is certainly not just a fluke.

thierryzoller, 2020-06-24:
Maybe signed/unsigned error?

supermerill, 2020-06-24:
i really don't know.

thierryzoller, 2020-06-26:
Happen sporadically, let's close this until we are able to reproduce

closed 2020-06-26 by thierryzoller

robthide37, 2020-07-17:
i can make this happen pretty consistently usually by editing line width and almost alway with editing individual settings for multiple items