supermerill/SuperSlicer #314: SS logo

Coffee0297, 2020-06-24:
Hello, i would strongly purpose to change the SS logo. its a real no go in German because of WW2. 

Kind regards Tonny Møller

supermerill, 2020-06-24:
Erf, that was a concern, hoping that the futuristic design won't convey that old problem.
What can i do?

supermerill, 2020-06-24:
I've updated a bit the acronym logo to have something less explicit.
But It's temporary, waiting to find something better.

n8bot, 2020-06-24:
Which logo are you both referring to? Logo design and branding is a skill I've been trying to learn about in the past few years, so I would love to play around if it may help. All I see for a logo is the blue Slic3r icon. Is there another logo I'm missing?

supermerill, 2020-06-24:
It's the "social medai preview".
[image]

As it's 640x320 and the slic3r icon is more tall than wide, i thought to add an acronym.

n8bot, 2020-06-24:
Is that logo found in the repository somewhere? I'm still confused as to where this logo exists, but I'll try and brainstorm some ideas in my spare time.

supermerill, 2020-06-24:
It's [hidden in a config menu](https://help.github.com/en/github/administering-a-repository/customizing-your-repositorys-social-media-preview) I'm the only one can access.

n8bot, 2020-06-24:
Ah yes, I understand what you mean now.

I use Adobe Illustrator, so I don't know if the Gimp file will be useful, but feel free to attach it if you feel it's important.

What aspects of the logo would you like to change and what aspects would you like to remain the same?

Are you happy with the font choice? Are you happy with the blue slic3r icon?

Do you just want the exact same font and icon as you have now, but composed differently?

I had an idea to try to use an open source font like Ubuntu for the logo. Would you object to that?

supermerill, 2020-06-24:
> What aspects of the logo would you like to change 

It shouldn't made someone think about some old german gang.

> and what aspects would you like to remain the same?

Big icon. Convey a bit the spirit of "more" or "super".

> Are you happy with the font choice?

sorta, wanted a sliced font, maybe a bit like the icon. This one is the best open-source i could find. I also thought about using the logo as the first S, but i don't have the hours to waste on that.

> Are you happy with the blue slic3r icon?

yes, it has been recently revamped. I also like the blue.

> I had an idea to try to use an open source font like Ubuntu for the logo. Would you object to that?

It has to be compatible with the agpv3

> Do you just want the exact same font and icon as you have now, but composed differently?

I want to include the blue slic3r icon (you can alter it a bit, but not too much).  The font can change. I don't mind if there's no text. It's just that the icon is a bit lonely in the banner. Also, including the name/acronym of the project make it more recognisable.

gdachs, 2020-06-24:
> Hello, i would strongly purpose to change the SS logo. its a real no go in German because of WW2.

It is not only because of WW2. We have still people in Germany that believes that this times should come back. The only way to treat them is to segregate them and show them that they are the scum of the earth.

That said, I believe SSlicer is not any better.

obelisk79, 2020-06-24:
Perhaps this font, with a similar glow effect as used in the example? The font name is also appealing for this application :)

n8bot, 2020-06-24:
> Perhaps this font, with a similar glow effect as used in the example? The font name is also appealing for this application :)
> https://www.dafont.com/sliced-tech.font?text=SS

Interesting font, but the license does not appear to be desirable for the application. It may work, but it's not clear.

As for the licensing of the Ubuntu font, here is the license: https://ubuntu.com/legal/font-licence

Is that compatible with AGPL3?

supermerill, 2020-06-24:
@obelisk79  that font isn't open-source.

Well, if everyone know that the first meaning of 'SS' is SuperSlicer, then the problem disappear! (yes, i know, it's not going to happen)

@n8bot  Yes, it's okay, there is no restriction on the usage, and I don't plan to bundle it.

n8bot, 2020-06-24:
> SSlicer is not better than the scum of the earth? outch.
> Or maybe i misunderstood.

I think he just means that having "SS" in any way shape or form is not a good solution. Though, I think you already addressed that for the immediate future with your revised logo.

In a few hours I'm going to try some variations, without going too far from what you have already.

n8bot, 2020-06-25:
I did a bit of work. I went crazy with a few wild ideas, but then I came back to reality. See first a few wild ideas:
[image]
[image]
[image]
[image]

So, then, I actually composed a few of the Social Media Preview logos, at 4x size:
[image]
[image]
[image]

And here are two glowing spheres at full resolution, one white BG, one transparent:
[image]
[image]

None of them are perfect, but I was just playing around. Let me know if you hate them or if you have any ideas for changes you'd like. Also, feel free to take any and all of these and use them as you wish.

It is Ubuntu Bold Italic font.

n8bot, 2020-06-25:
I played around with this some more this morning. I made the glow bigger, so that if the sphere is reduced to the size of an icon the glow is still visible. I also tried a black background lockup.

[image]
[image]

supermerill, 2020-06-25:
I'll use the flat design for now, I like it better than the other ones.

n8bot, 2020-06-25:
Very excellent choice. The outer glow was just an attempt to bring it closer to the original. I do also like the flat look better myself.

The font is still a little bit "playful" for me. It seems like you want to give the impression that SuperSlicer is cutting edge and aggressive. Do you think the text could use work?

I've done a few variations of the flat one for you for different scenarios.

[image]
[image]
[image]

Tell me to stop if you'd like me to stop flooding you with comments! Haha. For now, I don't have any other ideas to try out, but if anybody has a suggestion I will try it.

LaserBee, 2020-07-06:
How wed are you to the name "Super slicer"? I have to admit, I'm not a massive fan. It is a lot better than slic3r, because how the fuck do you pronounce that??

I don't know. Apart from the whole SS thing, "Super Slicer" doesn't seem like a great name. Would you be open to changing it entirely?

supermerill, 2020-07-07:
@LaserBee  it's the issue #78

closed 2020-10-29 by supermerill