supermerill/SuperSlicer #315: Release 2.2.52.0 does not compile

Lukas1818, 2020-06-24:
Release  2.2.52.0 and the last commit (27aa82c94fe01e89fbeeeb0909fcf6ddadd6b2cf) does not compile.
I get in both case the same error:
```
/home/lukas/git/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp: In member function ‘void Slic3r::GUI::FreeCADDialog::createSTC()’:
/home/lukas/git/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:636:18: error: ‘wxEVT_STC_AUTOCOMP_COMPLETED’ was not declared in this scope; did you mean ‘wxEVT_STC_AUTOCOMP_CANCELLED’?
  636 |     m_text->Bind(wxEVT_STC_AUTOCOMP_COMPLETED, &FreeCADDialog::on_autocomp_complete, this);
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  wxEVT_STC_AUTOCOMP_CANCELLED
/home/lukas/git/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:647:63: warning: conversion from ‘unsigned int’ to ‘wxColourBase::ChannelType’ {aka ‘unsigned char’} changes value from ‘256’ to ‘0’ [-Woverflow]
  647 |     m_text->StyleSetForeground(wxSTC_P_STRING, wxColour(128u, 256u, 0u)); // string, light green
      |                                                               ^~~~
/home/lukas/git/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:649:66: warning: conversion from ‘unsigned int’ to ‘wxColourBase::ChannelType’ {aka ‘unsigned char’} changes value from ‘256’ to ‘0’ [-Woverflow]
  649 |     m_text->StyleSetForeground(wxSTC_P_CHARACTER, wxColour(128u, 256u, 0u));
      |                                                                  ^~~~
/home/lukas/git/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp: In member function ‘void Slic3r::GUI::FreeCADDialog::test_update_script_file(std::string&)’:
/home/lukas/git/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:725:27: warning: catching polymorphic type ‘class std::exception’ by value [-Wcatch-value=]
  725 |     catch (std::exception ex) {
      |                           ^~
In file included from /usr/include/boost/asio/read.hpp:1288,
                 from /usr/include/boost/asio.hpp:112,
                 from /home/lukas/git/SuperSlicer/build/src/slic3r/libslic3r_gui_pch/pchheader.hpp:64:
/usr/include/boost/asio/impl/read.hpp: In instantiation of ‘class boost::asio::detail::initiate_async_read_dynbuf_v1<boost::process::detail::posix::async_pipe>’:
/usr/include/boost/asio/impl/read.hpp:834:15:   required from ‘auto boost::asio::async_read(AsyncReadStream&, DynamicBuffer_v1&&, CompletionCondition, ReadHandler&&, typename std::enable_if<(boost::asio::is_dynamic_buffer_v1<typename std::decay<_Tp2>::type>::value && (! boost::asio::is_dynamic_buffer_v2<typename std::decay<_Tp2>::type>::value))>::type*) [with AsyncReadStream = boost::process::detail::posix::async_pipe; DynamicBuffer_v1 = boost::asio::basic_streambuf_ref<std::allocator<char> >; CompletionCondition = boost::asio::detail::transfer_all_t; ReadHandler = boost::process::detail::posix::async_out_future<p1, p2, Type>::on_success<boost::process::detail::posix::executor<boost::fusion::joint_view<boost::fusion::tuple<boost::process::detail::posix::exe_cmd_init<char>, boost::process::detail::posix::io_context_ref>, boost::fusion::filter_view<const boost::fusion::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, boost::process::detail::posix::pipe_in&, boost::process::detail::posix::async_out_future<1, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::process::detail::posix::async_out_future<2, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::asio::io_context&>, boost::process::detail::is_initializer<mpl_::arg<-1> > > > > >::<lambda(const boost::system::error_code&, std::size_t)>; typename std::enable_if<(boost::asio::is_dynamic_buffer_v1<typename std::decay<_Tp2>::type>::value && (! boost::asio::is_dynamic_buffer_v2<typename std::decay<_Tp2>::type>::value))>::type = void]’
/usr/include/boost/asio/impl/read.hpp:808:20:   required from ‘auto boost::asio::async_read(AsyncReadStream&, DynamicBuffer_v1&&, ReadHandler&&, typename std::enable_if<(boost::asio::is_dynamic_buffer_v1<typename std::decay<_Tp2>::type>::value && (! boost::asio::is_dynamic_buffer_v2<typename std::decay<_Tp2>::type>::value))>::type*) [with AsyncReadStream = boost::process::detail::posix::async_pipe; DynamicBuffer_v1 = boost::asio::basic_streambuf_ref<std::allocator<char> >; ReadHandler = boost::process::detail::posix::async_out_future<p1, p2, Type>::on_success<boost::process::detail::posix::executor<boost::fusion::joint_view<boost::fusion::tuple<boost::process::detail::posix::exe_cmd_init<char>, boost::process::detail::posix::io_context_ref>, boost::fusion::filter_view<const boost::fusion::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, boost::process::detail::posix::pipe_in&, boost::process::detail::posix::async_out_future<1, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::process::detail::posix::async_out_future<2, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::asio::io_context&>, boost::process::detail::is_initializer<mpl_::arg<-1> > > > > >::<lambda(const boost::system::error_code&, std::size_t)>; typename std::enable_if<(boost::asio::is_dynamic_buffer_v1<typename std::decay<_Tp2>::type>::value && (! boost::asio::is_dynamic_buffer_v2<typename std::decay<_Tp2>::type>::value))>::type = void]’
/usr/include/boost/asio/impl/read.hpp:850:20:   required from ‘auto boost::asio::async_read(AsyncReadStream&, boost::asio::basic_streambuf<Allocator>&, ReadHandler&&) [with AsyncReadStream = boost::process::detail::posix::async_pipe; Allocator = std::allocator<char>; ReadHandler = boost::process::detail::posix::async_out_future<p1, p2, Type>::on_success<boost::process::detail::posix::executor<boost::fusion::joint_view<boost::fusion::tuple<boost::process::detail::posix::exe_cmd_init<char>, boost::process::detail::posix::io_context_ref>, boost::fusion::filter_view<const boost::fusion::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, boost::process::detail::posix::pipe_in&, boost::process::detail::posix::async_out_future<1, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::process::detail::posix::async_out_future<2, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::asio::io_context&>, boost::process::detail::is_initializer<mpl_::arg<-1> > > > > >::<lambda(const boost::system::error_code&, std::size_t)>]’
/usr/include/boost/process/detail/posix/async_out.hpp:136:32:   required from ‘void boost::process::detail::posix::async_out_future<p1, p2, Type>::on_success(Executor&) [with Executor = boost::process::detail::posix::executor<boost::fusion::joint_view<boost::fusion::tuple<boost::process::detail::posix::exe_cmd_init<char>, boost::process::detail::posix::io_context_ref>, boost::fusion::filter_view<const boost::fusion::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, boost::process::detail::posix::pipe_in&, boost::process::detail::posix::async_out_future<1, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::process::detail::posix::async_out_future<2, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::asio::io_context&>, boost::process::detail::is_initializer<mpl_::arg<-1> > > > >; int p1 = 1; int p2 = -1; Type = std::__cxx11::basic_string<char>]’
/usr/include/boost/process/detail/posix/executor.hpp:63:47:   required from ‘void boost::process::detail::posix::on_success_t<Executor>::operator()(T&) const [with T = boost::process::detail::posix::async_out_future<1, -1, std::__cxx11::basic_string<char> >; Executor = boost::process::detail::posix::executor<boost::fusion::joint_view<boost::fusion::tuple<boost::process::detail::posix::exe_cmd_init<char>, boost::process::detail::posix::io_context_ref>, boost::fusion::filter_view<const boost::fusion::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, boost::process::detail::posix::pipe_in&, boost::process::detail::posix::async_out_future<1, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::process::detail::posix::async_out_future<2, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::asio::io_context&>, boost::process::detail::is_initializer<mpl_::arg<-1> > > > >]’
[... cut ...]
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:334: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/FreeCADDialog.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1308: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:161: all] Error 2
```

Steps to reproduce:

```
mkdir -p build
cd build
cmake .. \
	-DCMAKE_BUILD_TYPE=Release \
	-DCMAKE_INSTALL_PREFIX=/usr \
	-DOpenGL_GL_PREFERENCE=GLVND \
	-DSLIC3R_FHS=ON \
	-DSLIC3R_STATIC=OFF \
	-DSLIC3R_WX_STABLE=ON \
	-DSLIC3R_GTK=3 \
	-DSLIC3R_BUILD_TESTS=OFF
make
```

Operating system: Linux 5.7.0-3-MANJARO

supermerill, 2020-06-24:
seems like your version of wxwidget is different as it doesn't have the define.
Did you rebuilt the dependencies? it's wxwidget 3.1 and not the old one?

msrd0, 2020-06-24:
I can reproduce this on ArchLinux, the latest available wxwidget version is 3.0.5.1 and the existence of the `SLIC3R_WX_STABLE` flag makes me assume that 3.1 is not (yet) stable.

Lukas1818, 2020-06-24:
I used 3.0.5.1-1. It is the newest version in the package sources.

supermerill, 2020-06-24:
I need 3.1 to have the wxStyledTextCtrl / scintilla

xzcvczx, 2020-07-14:
just to clarify wxEVT_STC_AUTOCOMP_COMPLETED was added in 3.1.1 https://docs.wxwidgets.org/trunk/classwx_styled_text_event.html

ollien, 2020-10-17:
Yep, I can reproduce this as well. I am on wxwdigets 3.0, which the docs say should be fine, but it seems that 3.1 has become a requirement

supermerill, 2020-10-17:
Yes, the doc is a bit out-of date.
WIll be updated on 2.3.54.

closed 2021-06-02 by supermerill