supermerill/SuperSlicer #317: 3DBenchy wheelhouse not attached

sshockley, 2020-06-25:
### Version
2.2.52

### Operating system type + version
Windows Server 2019 version 1809

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro, SKR E3 v1.2, Marlin 2.0.5.5

### Behavior
When slicing 3DBenchy in SuperSlicer the wheelhouse isn't attached to the deck except at the front.  Appears to work normally in PrusaSlicer 2.2.0.

_Is this a new feature request?_ No

sshockley, 2020-06-25:
[image]
[Project Files.zip](https://github.com/supermerill/SuperSlicer/files/4831480/Project.Files.zip)

supermerill, 2020-06-25:
It's the xy size compensation for hole the problem. Set it to 0 to go back to normal.
The problem is that it's set by default...

I just rewrote the code, and added some test to make this thing more reliable... seems like their is some cases not covered, so thank you to have found one!

sshockley, 2020-06-25:
Changing XY compensation to 0 did the trick.  Thanks!

closed 2020-08-24 by supermerill