supermerill/SuperSlicer #318: [FR] Clearer Guidance on Printer Settings & Seperate Printing Accell from Travel accell.

thierryzoller, 2020-06-27:
Printer Settings -> Machine Limits

1. Seperating Printing acceleration from "Retraction". M204 T is described as "Maximum Acceleration when retracting" - Can we make that "Maximum acceleration when travelling". Add to the description: Travel moves, includes retraction and any non printing move"
2. "Maximum Acceleration when extruding" is described as "M204 S". This should be M204 P
3. Can we add a dedicated  "Retract Acceleration" ? M204 R =

Overall: Can we align all of this to non legacy marlin ? https://marlinfw.org/docs/gcode/M204.html 
It appears I cannot seperatly configure Printing and Travel accell . Which can help reduce stringing (rampup Travel accell)

supermerill, 2020-06-28:
added the travel speed. It's only set for marlin firmware, and it's almost never used. But it's here.

thierryzoller, 2020-06-29:
My description may be not very clear, sorry for that. I was also pointing out that the descriptions/labels do not match the function, which leads to misunderstandings. Can this be aligned to 1. what really is happening in slicer 2. the correct nomenclature ?

thierryzoller, 2020-06-29:
> added the max travel speed. It's only set for marlin firmware, and it's almost never used. But it's here now.

Do you mean "travel accell" or speed?

supermerill, 2020-06-29:
acceleration, sorry.

correct nomenclature is the inverted 'R' and 'T', right? if yes, it's done.

thierryzoller, 2020-07-03:
I'll check once a new release is out :) Thanks

closed 2020-10-22 by supermerill