supermerill/SuperSlicer #319: [BUG] Modifier turns into part upon slicing

thierryzoller, 2020-06-27:
### Version
Latest release.

1. Load attached file
2. Inspect the Modifier , it is supposed to change speed
3. Slice it, and the modifier turns into part of the model
[modifierbug_2.zip](https://github.com/supermerill/SuperSlicer/files/4840579/modifierbug_2.zip)
[image]

supermerill, 2020-06-27:
Modifiers that works on perimeters have to  create "regions", one per setting. Then, the perimeters are printed inside each regions.
By doing that, you have created an other region, and then you print perimeters over these two regions.

Ideally, speed doesn't need to do that, but it's not possible int he current architecture.

thierryzoller, 2020-06-28:
So there is no way to change speed for distinct layers this way?

supermerill, 2020-06-28:
you can use it with the height modifier, it won't create these things.
But it's not possible to have distinct speed on the same layer the way you want.

closed 2020-08-25 by supermerill